commaai/openpilot #625: Question about stop and go.

njbrown09, 2019-04-30:
On cars with 0mph longitudinal, does that mean i can engage openpilot at a standstill? I noticed toyotas need to be at last 20 mph for Full speed DCC. Does 0mph longitudinal control remove this limitation?

zorrobyte, 2019-05-09:
Enjoy:
https://github.com/zorrobyte/openpilot/commit/693e10dfff9f66a7c2c62e15f93c71fc76e3e551

Note that OP can't send accelerator commands if the v_Ego is 0. It can only release the brakes, allow the car to "creep" forward, then OP can send accelerator commands starting at 1MPH. This is why it's not enabled, those pesky edge cases of the car not creeping forward on hills.

EasternPA, 2019-05-18:
Note that there are some cars where OP does not have longitudinal control. In my Accord, for instance, when I come to a stop manually, I can put the Brake Hold on and activate OP while sitting at a red light. Then, when the light turns green, I'll either tap the gas pedal to get the car moving or -- depending on the version I'm running -- hit the gas pedal and immediately hit resume on the cruise control to activate OP with the settings I set while sitting at the red light. I can't activate it while sitting with my foot on the brake pedal but I absolutely can when brake hold is holding the brakes for me.

When I manually stop behind traffic (the car will not stop when approaching vehicles stopped in your lane), I will often stop a full car length early and then reactivate OP. That causes my car to creep or lurch forward and stop right away as it expects to get going again only to see a car stopped a few feet ahead. But that's technically not activating OP at a standstill like you asked. That's with my foot off the brakes so it may be creeping for a split second.

pd0wm, 2019-06-02:
Looks like your question is answered. Feel free te reopen when you have more questions.

closed 2019-06-02 by pd0wm