commaai/openpilot #645: Create an issue template for sane creation of issues with most commonly needed information

zorrobyte, 2019-05-16:
**Notes for successful issue creation (delete this section after reading)**
Please title your issue as: OP Version - Make - Model - short description of the issue.

Example:
0.5.12 - Toyota Prius Prime Advanced - OEM self park functionality is terrible

/*OP Repo Maintainers, it would be incredible to have tags for this, such as:
OP Version
Make
Needs Replication
Replicated*/

# Description

A concise description of what the issue is.

## Expected behavior

A clear and concise description of what you expected to happen.

## Actual behavior

What is actually happening.

## To Reproduce

Steps to reproduce the behavior:
1. 
2. 
3. 

## Troubleshooting / Investigation details

What have you tried already? Have you attempted any codebase modifications? If so, please share a link to a (private) gist, or PR. 

## Cabana Explorer Segment ID (please attempt sharing the segment where the issue can be readily observed, scrub video to section then "Copy Current Segment"), Cabana Link
Segment ID, Cabana share link

theantihero, 2019-08-14:
We definitely need this to become a thing +1

ErichMoraga, 2019-08-14:
We already had a page for this stuff, but... this would probably be better.

ErichMoraga, 2019-08-16:
BTW - The page I was referring to is cited by the wiki...
https://community.comma.ai/wiki/index.php/Openpilot_Events

vanillagorillaa, 2019-10-10:
I know you know about this, but,  https://github.com/commaai/openpilot/blob/devel/.github/ISSUE_TEMPLATE/bug_report.md 

closing

closed 2019-10-10 by vanillagorillaa