commaai/openpilot #655: Testing openPilot on Windows 10

susan987321, 2019-05-21:
Hi, I am running openpilot on my PC using Windows 10 pro, by using
./run_docker_tests.sh
and get the feedback
...
Successfully tagged tmppilot: latest
Security Warning: You are building a docker image from windows against a non-windows docker host. all files and directories added to build context will have '-rwxr -xr -x' permissions. It is recommended to double check and reset permissions for sensitive files and directories.

Then the window will disappear. And nothing will appear in selfdrive/test/tests/plant/out, even the folder out is not created.

It seems the code  "$(pwd)"/selfdrive/test/tests/plant/out:/tmp/openpilot/selfdrive/test/tests/plant/out \
  tmppilot /bin/sh -c 'cd /tmp/openpilot/selfdrive/test/tests/plant && OPTEST=1 ./test_longitudinal.py'
did not be executed.

has someone got the same situation?

pd0wm, 2019-06-02:
Without a proper traceback or error message this will hard to debug. Maybe people will be able to help if you post that.

susan987321, 2019-06-03:
Thank you for your response. I have solved the problem I posted before by updating the setting of Docker for WSL. Now I got another problem. I tried to run test_longitudinal.py under WSL, and I failed.

Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named opendbc
make: Nothing to be done for 'all'.
make: Nothing to be done for 'all'.
ECU Camera Simulated: True
ECU Gas Interceptor: True
preimporting selfdrive.controls.radard
preimporting selfdrive.controls.controlsd
make: Nothing to be done for 'all'.
E
======================================================================
ERROR: setUpClass (__main__.LongitudinalControl)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./test_longitudinal.py", line 298, in setUpClass
    manager.prepare_managed_process('controlsd')
  File "/d/openpilot1/openpilot/selfdrive/manager.py", line 202, in prepare_managed_process
    importlib.import_module(proc)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/d/openpilot1/openpilot/selfdrive/controls/controlsd.py", line 22, in <module>
    from selfdrive.controls.lib.latcontrol import LatControl
  File "/d/openpilot1/openpilot/selfdrive/controls/lib/latcontrol.py", line 5, in <module>
    from selfdrive.controls.lib.lateral_mpc import libmpc_py
  File "/d/openpilot1/openpilot/selfdrive/controls/lib/lateral_mpc/libmpc_py.py", line 30, in <module>
    libmpc = ffi.dlopen(libmpc_fn)
  File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 141, in dlopen
    lib, function_cache = _make_ffi_library(self, name, flags)
  File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 802, in _make_ffi_library
    backendlib = _load_backend_lib(backend, libname, flags)
  File "/usr/local/lib/python2.7/dist-packages/cffi/api.py", line 797, in _load_backend_lib
    raise OSError(msg)
OSError: cannot load library '/d/openpilot1/openpilot/selfdrive/controls/lib/lateral_mpc/libmpc.so': /d/openpilot1/openpilot/selfdrive/controls/lib/lateral_mpc/libmpc.so: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called '/d/openpilot1/openpilot/selfdrive/controls/lib/lateral_mpc/libmpc.so'

Could you give some hints? Thank you!

pd0wm, 2019-08-16:
`libmpc.so` was precompiled for aarch64, and therefore didn't load. We no longer ship precompiled binaries in `devel` anymore, so this should no longer be a problem.

closed 2019-08-16 by pd0wm