commaai/openpilot #678: Unable to run user-created SNPE models on the EON

ShaneSmiskol, 2019-06-02:
I've been trying to figure out a way to run my SNPE 1.25 converted model alongside visiond's 1.19 however it seems to be impossible, as you can't load two different shared libraries with the same name and use them at the same time. I can load libSNPE.so 1.25 and run my custom model, however this breaks visiond and the models that comma has created. Vice versa for running 1.25 models on SNPE 1.19.

Would it be possible for comma to provide a download link for SNPE 1.19, or for comma to updgrade its models to 1.25? Under the revision history, it looks like Qualcomm has increased efficiency in the newest model format. If releasing the entire SNPE SDK isn't possible, how about the `snpe-tensorflow-to-dlc` executable?

This would be a huge help to the development community of openpilot. Thanks

pd0wm, 2019-06-02:
For 0.6 I'm building a fresh copy of NEOS, with updates for most of our libraries. I'm pretty sure that will include an update to SNPE 1.25.

In the meantime I'm afraid there is not much we can do. The SNPE licence does not allow redistribution. They even went as far as making a list with files they approve to distribute, but that only includes .so files.

ShaneSmiskol, 2019-06-03:
Awesome to hear! By the way, for anyone with my problem in the future; Qualcomm keeps an archive of old SNPE versions on their website hidden away. No idea how I couldn't find this for weeks, but here it is: https://developer.qualcomm.com/software/qualcomm-neural-processing-sdk/tools

pd0wm, 2019-06-04:
I assume they put it their after you requested it on the forum.

Closing this issue since it has been resolved.

closed 2019-06-04 by pd0wm