commaai/openpilot #681: Use Snapdragon LLVM compiler

jqnatividad, 2019-06-02:
https://developer.qualcomm.com/software/snapdragon-llvm-compiler-android/tools

With optimization flags
`APP_CFLAGS="-O3 -mcpu=cortex-a57"`

https://developer.qualcomm.com/qfile/26953/readme-install-snapdragon-llvm-609.pdf

And do some minor refactoring of C code to squeeze more performance

https://www.slideshare.net/QualcommDeveloperNetwork/22-snapdragon-llvmbenaissa918gg45

pd0wm, 2019-06-02:
Currently there is no cross-compilation support for the binaries in openpilot, so everything is compiled on the phone. Currently we're using clang 6, and we will soon upgrade to clang 8.

I'll look into the -mcpu flag, maybe that provide some speedups.

jqnatividad, 2019-06-02:
The Snapdragon LLVM compiler license seems permissive enough.  

https://developer.qualcomm.com/node/66842?referrer=node/28592

Perhaps, you can run internal benchmarks first to see how big a speedup you get with it, and if it’s big enough (the sample recompilation examples they cite 20-30% speed ups), perhaps, distribute as a blob the Snapdragon LLVM compiler?

jqnatividad, 2019-06-02:
And it seems the Snapdragon LLVM compiler is being baked into EON NEON Builder

https://github.com/commaai/eon-neos-builder/commit/1b899b17010f9d713ca5c91619c6a9a1613b0e2f

pd0wm, 2019-06-04:
the snapdragon compiler only runs on PC (x86), so we can't bundle it and run it on the EON. If we want to switch to that compiler we need to be a able to cross-compile from the computer to the EON.

closed 2019-08-16 by pd0wm