commaai/openpilot #683: Update very old version of qpOASES library

jqnatividad, 2019-06-02:
Bundled version is 1.3
https://github.com/commaai/openpilot/blob/devel/phonelibs/qpoases/VERSIONS.txt

Latest version is 3.1
https://projects.coin-or.org/qpOASES/browser/trunk/VERSIONS

pd0wm, 2019-06-02:
Unfortunately version 1.3 is the only version supported by the code generation part of Acado, our Non-Linear Model Predictive Control library.

Unfortunately it looks like acado is no longer being developed. However, there is not much choice in NPMC libraries that can run real-time on embedded hardware. At some point I might decide to write a subset of acado functionality for use in openpilot, but that is a huge undertaking.

renegadephysicist, 2019-07-02:
FYI Acado does support the C version of qpOASES 3.1.

Just use:
mpc.set(QP_SOLVER, QP_QPOASES3);

pd0wm, 2019-07-02:
Good catch, I'll check it out!

closed 2021-02-03 by adeebshihadeh