commaai/openpilot #701: Honda Insight 2019

nagolinc, 2019-06-16:
I have a 2019 Honda Insight EX.  LKAS and ACC are both standard.  Will this work with the Honda Bosch, or what do I need to get started?

jasonhc73, 2019-07-30:
I have the Touring of the 2019 Honda Insight, it is not supported, yet.

csouers, 2019-07-30:
Energee had a port for this. It has a conflicting fingerprint thus why it’s not upstreamed yet. I think kegman also has a fork for it.

dustinfields, 2019-07-31:
@nagolinc @jasonhc73 

We have an insight group with a working port
Join discord and add me as a friend ill invite you into the group

dustinfields

nagolinc, 2019-09-06:
@dustinfields  Not super familiar with Discord.  It says I need a "four digit tag" to add you

dustinfields, 2019-09-06:
Ironworm#8498

⁣Get BlueMail for Android ​

rbiasini, 2019-09-11:
Discussion moved to discord. For support we'll need a VIN based fingerprint.

closed 2019-09-11 by rbiasini