commaai/openpilot #710: Disable lane keeping during lane change + decrease it during manual steer

pratikchhatbar, 2019-06-21:
If I am showing right (left) turn side light, I plan to change lane to the right (left). Wouldn't it be easy to disable lane keeping torque on the steering wheel when the following two conditions are met?
1. Turn signal is ON
2. Steer torque in the direction of the turn signal
And then enable lane keeping when the following condition is met?
- No torque on the steering wheel by the driver
This would ease out lane changes! 

Additionally, the following would be helpful to prevent car from bumping on the pot holes: 
- Decrease the lane keeping control gain when manual torque is detected on the steering wheel.

icmma, 2019-06-24:
it has been done before

theantihero, 2019-08-14:
check out kegman's branch he has a simple way of disabling lkas when blinkers are on

nojvek, 2019-10-08:
To be honest. Openpilot needs more switches in its settings. 

There’s a whole bunch of really cool features that the community is working on and it’s in different branches

What we need is feature flags. By default openpilot is in safe mode. But you can enable certain feature flags via settings to try out “beta” features like automatic lane change, park assist, stop and go, red light detection, aggressive braking, acceleration and turning etc.

ErichMoraga, 2019-10-08:
@nojvek Most people "need" safety prioritized over features.  Also, *fewer* choices promote a consistent, straightforward user experience.

HaraldSchafer, 2020-01-20:
With lane changes shipped this is no longer relevant

closed 2020-01-20 by HaraldSchafer

nojvek, 2020-01-20:
Whoa! so excited for automatic lane changes.