commaai/openpilot #719: Images for OpenStreetMap project

RobJN, 2019-07-01:
comma.ai was mentioned at our recent OpenStreetMap event. In short OpenStreetMap is the wikipedia of map data - a crowd source project to build a map database all openly licenced. To do this we need to survey the world and having access to street level imagery helps us. 

There are two projects which collect street level imagery which can be used to help edit OpenStreetMap. These are [OpenStreetCam](https://openstreetcam.org) and [Mapillary](https://www.mapillary.com/).

Both the existing options are great but there are a LOT of roads to cover - it would be amazing if comma.ai users could support OpenStreetMap with imagery too.

Let me know if this is of interest - it seems to align with the open source nature of this project but I'm new here so may have misjudged it entirely.

geohot, 2019-07-01:
I'm not opposed to it, I'd love if our users had the option to contribute. But I'm not sure if we have the engineering resources to write or support this.

Fortunately, openpilot is MIT licensed and has a thriving fork community. This is a perfect feature for a fork, write it and the users will come.

closed 2019-07-01 by geohot

arne182, 2019-07-02:
@RobJN, I was planning something like this. Am busy using OSC to make my OSM edits atm.

RobJN, 2019-07-02:
@arne182 : Feel free to have a go at it. I'm not able to code this myself but have passed the idea around the OpenStreetMap community. No takers yet. If you're looking at OSC then maybe speak with @mvexel to see if Telenav can throw any help in.

mattbsea, 2020-07-30:
Hey @RobJN - did anyone ever follow up on this?  

Comma has an API here:  https://api.comma.ai/#comma-api-spec

I was thinking about a web service that OpenPilot users could log into that would let them authorize Comma API access and OSM API access.  Then grab GPS traces and imagery from their comma account and upload to OSM.  Once authorized, it would run nightly/weekly to download batches of new data and upload to OSM.

Would that be useful to OSM/OSC?  Feel free to reach out me on the comma discord (mattbsea) if you want to discuss this more.

RobJN, 2020-07-30:
Hi @mattbsea. Nobody took this forward as far as I'm aware. GPS traces and Street level images are all hugely beneficial for OpenStreetMap. The GPS traces can be pushed direct to OSM, but for images OSM relies on hosting from either OSC or Mapillary.

What you propose sounds sensible to me but I should state that I'm not a developer. I'd been more that happy to put you in touch with relevant people though if you do want to take this forward.

Thank you
Rob

mattbsea, 2020-08-06:
Hey @RobJN, I've started working on this.  I'm implementing it as a system tray application for Windows/Mac/Linux and also as a docker app that users run on their laptop/computer for just their own data.  I joined the OSM World discord and sent you a friend request at "RobJN#4996".  I also joined the Slack workspace: osmus.slack.com  Or email me directly at mbarclay <at> pm <dot> me. Let's continue discussions off the OpenPilot issue tracker.