commaai/openpilot #724: LKAS without ACC

kavalpatel23, 2019-07-02:
I was wondering if it would be possible for OpenPilot to maintain lane keep but disengage cruise control. Sometimes in traffic, I'd like to maintain the speed and braking, but have OpenPilot assist with lane keeping. Without OpenPilot, in my Honda CRV, if I press the LKAS button, the car starts assisting with lane keep, but it does not engage cruise. I was wondering if something similar would be possible with openpilot.

icmma, 2019-07-03:
anything is possible with code

Martint1980, 2019-07-23:
Please add it to a new version. Also that OP doesn't disengage when push the acceleration pedal.

ErichMoraga, 2019-07-23:
Not gonna' happen.  @geohot has mentioned this many times.  You can try Arne's fork, though.

arne182, 2019-07-23:
Arne to the rescue.
https://github.com/arne182/openpilot

ErichMoraga, 2019-07-23:
Yeahhh!

closed 2019-07-26 by kavalpatel23