commaai/openpilot #742: Network information not available in 0.6.0

deftdawg, 2019-07-14:
I can't find the networking info (what IP address, etc the EON has) anywhere within the EON's UI.

Without that it's makes it difficult to connect to the EON when trying to connect to it over a phone hotspot connection.

ErichMoraga, 2019-07-14:
You've asked several questions via GitHub that have already been answered in Discord.  Nothing has changed in 0.6 regarding how to retrieve the IP address.

theantihero, 2019-07-17:
There's custom UIs that lets you do this. Check out kegman's or arne's repos.

ErichMoraga, 2019-07-17:
Exactly, that's why we have custom repos.  Just because your IP is apparently changing all the time, and you want to see it without drilling into a couple menus, doesn't mean the masses need it plastered on the UI (which Comma generally wants to keep clean).

rbiasini, 2019-09-11:
IP address is available from `Advanced Settings->Networks->Open WiFi Networks->Advanced`

closed 2019-09-11 by rbiasini