commaai/openpilot #743: Bluetooth tethering not available in 0.6.0

deftdawg, 2019-07-14:
Prior to 0.6.0, I would tether my EON to my phone over Bluetooth.  

This arrangement would allow me to SSH from my phone to the EON while in the car, no where close to a wifi network, while still allowing my phone to stay on cellular.

With 0.6.0, the Bluetooth settings button has disappeared, and so I can no longer pair with my phone and select it for "Internet".

Please advise how I can work around this limitation.

rbiasini, 2019-09-11:
For power saving reasons and little use case for it, Bluetooth was disabled in 0.6.

arne182, 2019-09-11:
Another reason to downgrade to 0.5. could you at least supply a kernel where it is enabled?

ErichMoraga, 2019-09-11:
@deftdawg presented an interesting use case.  Another use case includes using keyboard/mouse, which I've found useful on occasion.  Regardless, Bluetooth only consumes power when enabled @rbiasini

arne182, 2019-09-11:
I petition that it be readded and disabled by default.

closed 2020-01-20 by pd0wm