commaai/openpilot #744: Nano editor no longer available in 0.6.0

deftdawg, 2019-07-14:
With 0.6.0, the nano text editor is no longer available.   It is not installed, when attempting to run it, you are advised to run "pkg install nano", however when you run "pkg install nano" you are told you "MUST not run this command as root".   

How can I re-enable nano?

rjschof, 2019-07-21:
Suck it up and use a real text editor like vim

ErichMoraga, 2019-07-21:
Agreed!  If necessary, @deftdawg can follow the installation instructions I pinned in Discord.

deftdawg, 2019-07-21:
Cool, which channel did you pin the install instructions or how to fix the broken pkg system in?

ErichMoraga, 2019-07-21:
I pinned it 7/1 in #openpilot ... and when you SPAM'd the channels about it 7/14, @vanillagorilla responded in #help that it would be fixed (I pinned that post a few minutes ago, as a reminder).

deftdawg, 2019-07-21:
Found it thanks!

nano is also gone, but you can reinstall: 
```bash
mount -o remount,rw /system && apt update && apt install nano
 ```

closed 2019-07-21 by deftdawg