commaai/openpilot #750: Unusable  .6 load on 2018 Camry

mikecboyd, 2019-07-20:
Idk who touched the code but good god this load is unusable. Severe pull to the right anytime it is engaged rendering unusable. Prior load did not have this issue.

ErichMoraga, 2019-07-20:
Known issue.  Start your car with the wheels straight.

illumiN8i, 2019-07-21:
The fix is in the devel branch now.

ErichMoraga, 2019-07-21:
Yuuup!  Saw that this morning.  Hopefully that'll keep the kiddos happy until Christmas!

ErichMoraga, 2019-07-21:
@mikecboyd https://github.com/commaai/openpilot/pull/751

mikecboyd, 2019-07-21:
Awesome, will wait for release. Estimates on when next one will be?

ErichMoraga, 2019-07-21:
It's already in devel.

mikecboyd, 2019-07-21:
How do I point install to the devel? I’m used to just typing in the normal link for install when needed.

ErichMoraga, 2019-07-21:
Please check Discord and the wikis, since you might have follow up questions, and this isn't really the right place for them.

rbiasini, 2019-07-22:
fix merged both in devel and release2.

closed 2019-07-22 by rbiasini