commaai/openpilot #754: Supported cars list is wrong, 2020 Toyota Corolla is not fully supported

meritus, 2019-07-22:
I found out the 2020 Toyota Corolla LE is not fully support even though it says under supported cars: "All". Found this out after getting the vehicle. Please fix.

rbiasini, 2019-07-22:
This is most likely a fingerprint issue, i.e. your vehicle might generate CAN messages that we have not seen before on that vehicle model. This is always a possibility on every vehicle model.

You can join dicord.comma.ai for a more in-depth discussion.

meritus, 2019-07-23:
There has been discussion on this, there doesn't seem to be support for the 2020 Corolla LE out of the box and trying to run get_fingerprint.py does not yield anything useful.

[image]

rbiasini, 2019-09-11:
@meritus do you still have issues?

closed 2019-09-11 by rbiasini