commaai/openpilot #760: how to understand the vanishing point formula

gucasbrg, 2019-07-25:
```
new_vp = eon_intrinsics.dot(view_frame_from_device_frame.dot(trans))
```
trans means speed? can you provide some article to explanation  these formula

dekerr, 2019-07-30:
trans as in translation of the camera frame. 
this is probably a decent start to understand frames and camera properties
http://ksimek.github.io/2013/08/13/intrinsic/

gucasbrg, 2019-08-07:
@dekerr 
> ```python
> def handle_cam_odom(self, log):
> trans, rot = log.cameraOdometry.trans, log.cameraOdometry.rot
> if np.linalg.norm(trans) > MIN_SPEED_FILTER and abs(rot[2]) < MAX_YAW_RATE_FILTER:
>   new_vp = eon_intrinsics.dot(view_frame_from_device_frame.dot(trans))
> ```

the defination of  **trans** in  CameraOdometry,looks like  **m/s in device frame**

>
> ```
> struct CameraOdometry {
>   trans @0 :List(Float32); # m/s in device frame
>   rot @1 :List(Float32); # rad/s in device frame
>   transStd @2 :List(Float32); # std m/s in device frame
>   rotStd @3 :List(Float32); # std rad/s in device frame
> }
> 
> ```

gucasbrg, 2019-08-27:
[image]

closed 2019-08-27 by gucasbrg