commaai/openpilot #771: EON appears to have panic'd during the middle of drive

billma12, 2019-07-31:
Sorry not sure where else to address this besides github issues.

Yesterday, was cruising like 75mph on highway.

Then openpilot disengaged and I got a bunch of warnings on my dashboard. Which was weird because EON Was still green.

But actually I realized EON froze (and later rebooted itself)

https://my.comma.ai/cabana/?route=cb38ea752c72da81%7C2019-07-30--10-10-47&max=16&url=https%3A%2F%2Fchffrprivate.blob.core.windows.net%2Fchffrprivate3%2Fv2%2Fcb38ea752c72da81%2Ffd2cb58b6d81c4f0b71167bc3b6b5a66_2019-07-30--10-10-47

This is the link I found from Cabana that shares data from the drive. Time of interest would be at the end when the system disengaged itself.

Let me know if need more information or there's a better place to post this issue.
Been using openpilot for like 6 months without any hiccups until this happened.

Thanks.

Build: 6.1 dev 
Car: 2017 Toyota Corolla
Pedal: Yes

cydia2020, 2019-09-08:
I've ran into this issue once before, OpenPilot 0.6 was running at the time, Toyota Prius 2018

bugsy924, 2019-09-08:
Any chance your EONs are overheating?

ErichMoraga, 2019-09-08:
I'm sure everyone has had an issue at some point, but considering there've been several NEOS revisions in the last few months, please upgrade to the latest... and we'll go from there if your issues persist.

cydia2020, 2019-09-09:
> Any chance your EONs are overheating?

No, it's bloody cold here in victoria😂

pd0wm, 2020-01-20:
Closing this issue now since there have been many stability improvements over the past months. Feel free to open a new issue if you still have random reboots/freezes.

closed 2020-01-20 by pd0wm