commaai/openpilot #778: Can this enhance the steer turning angle strongly?

gzjackar, 2019-08-03:
I find CRV Hybrid  can make the steer turn over 300 degrees when it do the auto-parking-assistant , how can I find the big torque DBC code to replace the original poor steer control code? Could someone help?
Here is the cabana link of 300 degrees turning :
https://my.comma.ai/cabana/?route=c9ea5789a10a55ac%7C2019-07-17--00-28-42&max=8&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fc9ea5789a10a55ac%2F8b6ce67f43d89cf810870807f6782c1c_2019-07-17--00-28-42
Here is the time point of auto-parking-assistant:
[image]
 Here is the auto-parking-assistant video:
https://cdn.discordapp.com/attachments/600490201197969418/600871699512295435/480-_2019-07-17_10.06.59.mov

Martint1980, 2019-08-03:
Please use it for op!

energee, 2019-08-05:
@gzjackar What country is this CR-V from?

csouers, 2019-08-05:
I have yet to find a command signal for this. Spent a good bit of time on it this weekend. Is the eps on a different can than the camera?

energee, 2019-08-05:
I was looking for it too without success, i mean EPS is def on the same bus since we see all the normal EPS rx and tx messages but I suppose its possible that there is another private bus between park assist and EPS.

energee, 2019-08-05:
Interesting that that bit is high on 0xe4, ive seen this before on other chinese hondas.

jyoung8607, 2019-08-05:
Before you get too excited about this path, bear in mind the EPS rack may have limitations on that control method. For example, Volkswagen and Audi EPS also supports angle-based control for Park Assist with inherently high torque, but the EPS rack itself will reject that style of control above 15kph.

energee, 2019-08-05:
Yes just to be clear, this control method is not very relevant to openpilot. Even with enough understanding about how it works, it will almost certainly never be supported as it likely requires current speed and possibly current gear to be spoofed. I believe @rbiasini would confirm that this falls outside of openpilot safety specifications.

csouers, 2019-08-05:
@energee I think the data we’re seeing might be forwarded. Will be looking at a diagram soon. This crv has 2-4 new frames around 0x183 that appear to have angle data, but it’s not command because it’s active with manual driving and is synced to the normal sensor. Doesn’t appear to be as high of resolution as the normal sensor signal either.

gzjackar, 2019-08-06:
> @gzjackar What country is this CR-V from?

China

gzjackar, 2019-08-06:
> @gzjackar What country is this CR-V from?

> I have yet to find a command signal for this. Spent a good bit of time on it this weekend. Is the eps on a different can than the camera?

Is this steer torque signal a key ?
[image]

gzjackar, 2019-08-06:
[image]

[image]

energee, 2019-08-06:
Steer_status (and steer_sensor) message is being read by EON from EPS module, Steering_control msg is being sent from EON to EPS, that is what these module references indicate on the DBC.

gzjackar, 2019-08-07:
> Steer_status (and steer_sensor) message is being read by EON from EPS module, Steering_control msg is being sent from EON to EPS, that is what these module references indicate on the DBC.

So what is the hidden command signal for big turn (Auto Park Assitant) ?  I have thought it is the SteerTorqueMotor .

csouers, 2019-08-07:
Steertorquemotor is what the motor is currently doing (think read only). We don’t know where the command is yet. Probably on a different bus. We might know soon.

gzjackar, 2019-08-07:
> Steertorquemotor is what the motor is currently doing (think read only). We don’t know where the command is yet. Probably on a different bus. We might know soon.

wow, It's very cool, hope to know soon.

pd0wm, 2020-01-20:
Closing this issue because of inactivity.

closed 2020-01-20 by pd0wm