commaai/openpilot #779: OpenPilot 0.6x can not go back to 0.5x?(I have tried many ways)

gzjackar, 2019-08-06:
Because the poor turning ability of steer turning (Honda CRV-Hybrid) I get by OpenPilot 0.6x,I need to go back to 0.5x,I have tried many ways:
1、Thanks to @arne182, downgrade NEOS with one command
2、Manual Instructions to flash back to v9 NEOS for downgrading back to 0.5.xx:（Kegman）
3、Download V9 NEOS and flash.py (official method)
I have downgraded the NEOS from 10 to 9 and panda from 1.4.1 to 1.2.1, but I still can not get the op works properly ,now the eon can show the car speed ,and the panda is also active , but the car will show ACC,LKAS error, openpilot can not be enable either. Do some know the root caused?
I get the error message below :
[image]
[image]
[image]
[image]

gzjackar, 2019-08-07:
I replug the panda , the miracle appears , issue is closed

closed 2019-08-07 by gzjackar

ErichMoraga, 2019-08-07:
Yup, just needed to be power cycled.