commaai/openpilot #782: variable associated to a variety of coordinate systems

gucasbrg, 2019-08-07:
1. what mean the defination of **dRel\yRel?** and is the following explanation correct?
[image]

2. in this project, associated to a variety of coordinate systems, **car to model\car to bb\pt to car\pt from car ex,** can you explain some principle?

3. Pathplanner ,Convert the MPC control result to zmp, then use ui.c to plot x, y to the method Car_space_to_full_frame **vec4 p_car_space = (vec4){{px, py, 0., 1.}}; How to convert to this format?** Car_space_projective can be seen as a world coordinate system (xw, yw, zw, 1), Then use the external parameter matrix extrinsic_matrix to get the camera coordinate system Ep Then use the internal parameter matrix to get p_image 

 I am very sorry to delay your time and look forward to your reply.

ErichMoraga, 2019-08-07:
See... https://github.com/commaai/openpilot/issues/27

gucasbrg, 2019-08-07:
> See... #27
[image]

I see the file in "car/*/radar_interface.py", in gm, the code
```
self.pts[targetId].dRel = distance # from front of car
# From driver's pov, left is positive
deg_to_rad = np.pi/180.
self.pts[targetId].yRel = math.sin(deg_to_rad * cpt['TrkAzimuth']) * distance		、
```
Seemingly contradictory

pd0wm, 2019-08-12:
Interesting find. Not exactly sure why it is done this way. Since the angles are small this is not really a problem though, if dRel is 10m and the angle is 10 degrees the error is only 3 cm.

closed 2019-08-27 by gucasbrg