commaai/openpilot #791: Communication Issue between processes

mtourangeau, 2019-08-23:
Since I upgraded to version 0.6, I have a error when I start the car. AHB, camera and Rador are faulty. When I engage OP, I get communication issue between processes. And OP never work. 

I have a Prius Prime and used OP since may 2018. 
Here is the cabana

https://my.comma.ai/cabana/?route=27d5c512260e17cc%7C2019-08-06--01-37-44&max=9&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F27d5c512260e17cc%2F3ab10437d4203a35724c687f3fb72855_2019-08-06--01-37-44

[image]

ErichMoraga, 2019-08-23:
Hey Mathieu!  Thanks for your past assistance w/ lateral tuning on the Prime.  We really need to get you using the ZSS sensor, as Primes drive perfectly now.  Assuming you're having issues with Comma's latest 0.6.3, you should really try re-loading code again.  After you clone it from scratch, run `make` which should trigger the NEOS upgrade, if necessary.

mtourangeau, 2019-08-30:
I can confirm that this is a panda issue. The firmware of the panda does not get updated. I'll try to redownload 0.6.3 and then connect to my car. Even if I have messages when I start my car, I'll wait and try to activate OP.

mtourangeau, 2019-08-30:
root@localhost:/data/openpilot/panda/board$ cd /data/openpilot/panda/board && make recover
PYTHONPATH=../ python -c "from python import Panda; Panda().reset(enter_bootloader=True)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/data/openpilot/panda/python/__init__.py", line 141, in __init__
    self.connect(claim)
  File "/data/openpilot/panda/python/__init__.py", line 186, in connect
    assert(self._handle != None)
AssertionError
make: [build.mk:45: recover] Error 1 (ignored)
sleep 1.0
"tools/dfu-util-aarch64" -d 0483:df11 -a 0 -s 0x08004000 -D obj/panda.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

tools/dfu-util-aarch64: Invalid DFU suffix signature
tools/dfu-util-aarch64: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08004000, size = 32720
Download        [                        Download        [=                       Download        [===                     Download        [====                    Download        [======                  Download        [=======                 Download       1[=========               Download       1[==========              Download       1[============            Download       1[==============          Download       1[===============         Download       2[=================       Download       2[==================      Download       2[====================    Download       2[=====================   Download       2[======================= Download       3[=========================] 100%        32720 bytes
Download done.
File downloaded successfully
"tools/dfu-util-aarch64" -d 0483:df11 -a 0 -s 0x08000000:leave -D obj/bootstub.panda.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

tools/dfu-util-aarch64: Invalid DFU suffix signature
tools/dfu-util-aarch64: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 9444
Download        [                        Download        [=====                   Download        [==========              Download        [================        Download        [=====================   Download        [=========================] 100%         9444 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
root@localhost:/data/openpilot/panda/board$ 

Still not working

mtourangeau, 2019-08-30:
I think I have the first generation of panda. Bought it in april 2018.

energee, 2019-08-30:
If you _bought_ your panda (Didnt receive it for free through the old contests), there is no difference in generation. All white pandas are Rev C, regardless of whether they are in 3d printed case or injection molded case.

mtourangeau, 2019-09-25:
replacing the panda fixed the problem.

closed 2019-09-25 by mtourangeau