commaai/openpilot #796: Compiling visiond PC version from #544

susan987321, 2019-08-29:
I am building and compiling visiond PC version from #544, But I got several errors as follows.
1>D:\VS2019Code\TestConsoleApplication1\cereal\gen\cpp\log.capnp.h(10,2): error : #error "Version mismatch between generated code and library headers.  You must use the same version of the Cap'n Proto compiler and library."
1>D:\VS2019Code\TestConsoleApplication1\cereal\gen\cpp\log.capnp.h(10,2): error :  #error "Version mismatch between generated code and library headers.  You must use the same version of the Cap'n Proto compiler and library."
1>D:\VS2019Code\TestConsoleApplication1\cereal\gen\cpp\log.capnp.h(10,2): error :   ^~~~~
1>D:\VS2019Code\TestConsoleApplication1\cereal\gen\cpp\car.capnp.h(10,2): error : #error "Version mismatch between generated code and library headers.  You must use the same version of the Cap'n Proto compiler and library."
1>D:\VS2019Code\TestConsoleApplication1\cereal\gen\cpp\car.capnp.h(10,2): error :  #error "Version mismatch between generated code and library headers.  You must use the same version of the Cap'n Proto compiler and library."
1>D:\VS2019Code\TestConsoleApplication1\cereal\gen\cpp\car.capnp.h(10,2): error :   ^~~~~

1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(233,14): error : ‘class cereal::FrameData::Builder’ has no member named ‘setLensPos’
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(233,14): error :        framed.setLensPos(frame_data.lens_pos);
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(233,14): error :               ^~~~~~~~~~
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(234,14): error : ‘class cereal::FrameData::Builder’ has no member named ‘setLensSag’
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(234,14): error :        framed.setLensSag(frame_data.lens_sag);
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(234,14): error :               ^~~~~~~~~~
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(235,14): error : ‘class cereal::FrameData::Builder’ has no member named ‘setLensErr’
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(235,14): error :        framed.setLensErr(frame_data.lens_err);
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(235,14): error :               ^~~~~~~~~~
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(236,14): error : ‘class cereal::FrameData::Builder’ has no member named ‘setLensTruePos’
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(236,14): error :        framed.setLensTruePos(frame_data.lens_true_pos);
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(236,14): error :               ^~~~~~~~~~~~~~
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(241,14): error : ‘class cereal::FrameData::Builder’ has no member named ‘setTransform’
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(241,14): error :        framed.setTransform(transform_vs);
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(241,14): error :  

1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(312,31): error : ‘class cereal::Event::Builder’ has no member named ‘initCameraOdometry’; did you mean ‘initCarControl’?
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(312,31): error :          auto posenetd = event.initCameraOdometry();
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(312,31): error :                                ^~~~~~~~~~~~~~~~~~
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(312,31): error :                                initCarControl

1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(418,45): error : ‘class cereal::LiveCalibrationData::Reader’ has no member named ‘getWarpMatrix2’; did you mean ‘getWarpMatrix’?
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(418,45): error :        auto wm2 = event.getLiveCalibration().getWarpMatrix2();
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(418,45): error :                                              ^~~~~~~~~~~~~~
1>D:\VS2019Code\TestConsoleApplication1\visiond_pc.cc(418,45): error :                                              getWarpMatrix

For the first mismatch error, I think it occurs because of the code in "log.capnp.h"
#if CAPNP_VERSION != 5003
#error "Version mismatch between generated code and library headers.  You must use the same version of the Cap'n Proto compiler and library."
#endif

I have checked my capnp version on the Ubuntu, it is 0.6.1. However, the "log.capnp.h" is from OpenPilot v0.2.7 because there is not the head file in the lastest Openpilot. I think it is the reason. But I don't know how to solve it.

Is there somebody catch the same problem?  Many thanks.

pd0wm, 2019-09-10:
Make sure you install capnp according to the instructions in `openpilot-tools`: https://github.com/commaai/openpilot-tools#setup

susan987321, 2019-09-11:
Thank you for your suggestion.@pd0wm
I found how to solve this problem.  In recent several versions, there is not a folder, gen, in the folder, cereal. We must use makefile in the folder, cereal to generate the folder, gen. After that, "log.capnp.h"  will be generated. The mismatch error will disappear.

closed 2019-09-11 by rbiasini