commaai/openpilot #799: Car no longer plays alert tone when Openpilot disables itself after error while driving

wjohnson191, 2019-09-03:
I've been using Openpilot with my 2018 Toyota RAV4 Hybrid for about a year now. Multiple times per long drive, Openpilot will disable itself and the EON screen will show a red "TAKE CONTROL IMMEDIATELY" error, usually with a description below such as "CAN Error" or "Communication Error". I've tried to solve this issue by replacing the Giraffe, the Panda, and the USB cable, but to no avail.

Before Openpilot v0.6.3, this error would be accompanied by my car making a loud and long beeping tone, indicating that I needed to take over immediately. However, as of v0.6.3, this error no longer causes the car to beep at all - the EON makes a relatively quiet chime (easily missed if music is playing), and that's it.

This lack of a loud alert is clearly a safety issue, given that this error means that Openpilot is no longer exercising either lateral or longitudinal control.

I also experience every 15 minutes or so a "Steering is temporarily unavailable" error, where Openpilot loses lateral control for about 3-5 seconds. This also used to be accompanied by a shorter alert tone from the car, but as of v0.6.3, that alert is also gone.

The car still does sound an alert for certain situations, such as when Openpilot is disabled due to manual braking. So, the capability to issue car alerts in the first place does not seem to be an issue.

Has anyone else experienced this lack of car alerts on serious errors as of v0.6.3? I strongly believe the car should still play loud alerts for serious errors.

I'd also of course appreciate any insight into why I might be experiencing these serious errors in the first place, but that's not the intent of this issue.

ErichMoraga, 2019-09-03:
Do you use Driver Monitoring?  SIM?  What's your DIP set to?

bugsy924, 2019-09-03:
I have yet to dig into the steering unavailable issue. I also experience it in the 2018 camry hybrid. Frequency can be anywhere between a few seconds to hours of driving, I haven't noticed a trending trigger. The red errors you're getting are likely a loose connection somewhere. You can bring sounds back by editing carcontroller to send sounds when openpilot throws an alert

wjohnson191, 2019-09-04:
I use driver monitoring, have a data SIM installed, DIP is set to all off according to the [Toyota flowchart](https://github.com/commaai/neo/blob/master/giraffe/toyota/toyota_giraffe_switch_flowchart.png), and I’m running stock Comma OP 0.6.3.

wjohnson191, 2019-09-04:
Has anyone seen cases where there’s a loose connection inside the EON Gold itself? That’s the only piece I haven’t replaced yet, and I’ve checked all connections carefully and redone them many times.

Thanks for the pointer on where to change this behavior. I’ve been trying to avoid forking so that I can still get automatic updates, but I may have to in order to fix this issue - the lack of beeps on disengagements just feels unsafe. 

And glad to hear you’ve also seen the steering issue. Any pointers for how to dig into debugging that? Would be interested in helping out.

ErichMoraga, 2019-09-04:
@wjohnson191 It's interesting you get it in multiple cars, but I wouldn't be quick to blame the EON hardware (where the USB PCB would be most suspicious).

Humor this..
 Disable DM, and remove the SIM.

rbiasini, 2019-09-11:
The EON error chime volume has been increased in 0.6.4. Do you think it's acceptably loud, @wjohnson191 ?

ErichMoraga, 2019-09-11:
@rbiasini I tried 0.6.4 an hour or so ago, and found it shockingly quiet.  I was actually tempted to look at the media w/ studio software, check on the amplitude, etc.

wjohnson191, 2019-09-11:
@ErichMoraga I'll remove the SIM and disable DM and see if that helps with the underlying issue, thanks for the tip.

@rbiasini I'll try 0.6.4 on my next long drive. However, I remain strongly of the opinion that the car should beep upon serious errors, perhaps in addition to the EON playing a chime. I see no good reason not to alert the driver by any means possible upon serious errors.

ErichMoraga, 2019-09-11:
@wjohnson191 Don't forget DM can't be disabled in 0.6.4 ;-)

carmoses, 2019-10-13:
> @ErichMoraga I'll remove the SIM and disable DM and see if that helps with the underlying issue, thanks for the tip.
Have you had any success?

wjohnson191, 2019-10-16:
Yes - removing the SIM fixed all of the intermittent errors I described above. Thanks for the tip @ErichMoraga.

closed 2020-01-20 by pd0wm