commaai/openpilot #800: Update to SNPE 1.25?

ShaneSmiskol, 2019-09-04:
This issue was opened before 0.6.0 came out, however it was closed because SNPE was supposed to be updated when 0.6.0 came out.

There seems to be a few performance improvements as well as support for more TensorFlow operations/layers.

pd0wm, 2019-09-04:
Unfortunately the newer SNPE doesn't seem to be compatible with the Android version/build we use. After a certain version they started building it with clang instead of gcc, which seems to be the issue. It just segfaults when you try to use it.

Would be great if you would be able to get it to run!

ShaneSmiskol, 2019-09-05:
@pd0wm SNPE 1.25 or above? I've gotten 1.25 to run fine on the EON before.

pd0wm, 2019-09-06:
I think I tried 1.25 before, but I'll try again if you got it to work.

pd0wm, 2019-09-06:
Ok, looked into it again. Before we were using the clang3.8 version from SNPE 1.19.2 (`./lib/aarch64-android-clang3.8/libSNPE.so`) In 1.25.1 they switched to clang 6.0, that one segfaults when you try to load it. The GCC versions don't compile at all on the EON, I guess because we use clang on the EON to compile visiond.

Which exact .so files did you get to work on your EON?

ShaneSmiskol, 2019-09-17:
@pd0wm I'm not too sure, I just know I once downloaded SNPE 1.25 a while ago, the latest at the time, extracted the libs to run on a phone and got SNPE 1.25 to make predictions on my custom model.

ShaneSmiskol, 2019-09-17:
I can try the process again, if it would help you get 1.25+ working with openpilot!

pd0wm, 2019-09-17:
Yes, that would be great! Let me know exactly what steps you took to get it to work.

ShaneSmiskol, 2019-09-17:
I've been unable to get it working so far today (I deleted my old branches had had the code in the history:( ), but I found an old screenshot showing it can be done: https://cdn.discordapp.com/attachments/577201235896303617/582199349879111681/Screenshot_227.png

ShaneSmiskol, 2019-09-26:
@pd0wm Good news actually! Got 1.25 on the EON and it's predicting on a 1.25 model perfectly fine.

ShaneSmiskol, 2019-09-26:
[image]

https://github.com/ShaneSmiskol/openpilot/tree/snpe-v1.25.1.310-demo

All I did was leave all SNPE shared libraries except libSNPE.so from `aarch64-android-clang6.0`

ShaneSmiskol, 2019-09-26:
> @pd0wm Got SNPE 1.25 working! All I had to do was copy libSNPE.so from `aarch64-android-clang6.0` to the LD_LIBRARY_PATH and it works perfectly with this 1.25 model. Probably could work on later versions, I just haven't tested it yet.
> 
> [image]
> 
> https://github.com/ShaneSmiskol/openpilot/tree/snpe-v1.25.1.310-demo

geohot, 2020-01-23:
We have newer SNPE working, the sad thing is it uses more CPU to run the model. May revisit in the future, but for now, the version we have is working.

closed 2020-01-23 by geohot