commaai/openpilot #818: Any chance for supporting the 2014 Mazda 3 iTouring car?

shawneric, 2019-09-25:
I have a 2014 Mazda 3 iTouring sedan that I'm interested in setting up for self-driving, but I can't afford to buy another car at the moment. Any chance this car can be added to your list? I do know it has sonar and blind spot monitoring.

ErichMoraga, 2019-09-25:
Support isn't added just by putting it on the list.  Join the Comma Discord community, and have someone in the #mazda channel port it.  A pull request of that supporting code would have to be submitted/accepted to make "support" official.

vanillagorillaa, 2019-09-27:
Without ACC and LKAS this car can't be supported. I don't see this car as having those features. Closing, any further discussion can be on Discord as Erich suggested.

closed 2019-09-27 by vanillagorillaa

ryleymcc, 2022-04-11:
> I have a 2014 Mazda 3 iTouring sedan that I'm interested in setting up for self-driving, but I can't afford to buy another car at the moment. Any chance this car can be added to your list? I do know it has sonar and blind spot monitoring.

Done.