commaai/openpilot #822: Head light system malfunction after flooring the acceleration pedal

cydia2020, 2019-09-29:
Car: Toyota Prius 2018, ZVW50
OpenPilot version: 0.6.x (all experiencing this)

Steps to reproduce the problem:
1. Disengage OpenPilot
2. Stop the car
3. Floor the acceleration pedal
4. If you're lucky, you get a headlight system malfunction error on the MID

Resolution:
Restart the car

Note: this doesn't happen all the time, it only happens on very rare occasions, but it does happen and I can experience it around once a fortnight.

illumiN8i, 2019-09-29:
I had the same thing happen once when braking hard on 2018 Prius Prime. I believe the sudden deceleration caused the connection to the lane keep camera to come slightly lose. Make sure all your connections are solid.

arne182, 2019-09-29:
I agree. Loose connection.

cydia2020, 2019-09-29:
Thanks for the suggestion, I'll go ahead and make sure the connection is solid.

Sincerely
Cheng Chi

ErichMoraga, 2019-09-30:
I hate to break it to you, but I believe this is the bug I've ranted about a few times in Discord over the last few months affecting us when we use CAN forwarding with the DSU disconnected.

illumiN8i, 2019-09-30:
I have an example of it occurring if it helps at all. b29e3bc918751697|2019-08-10--18-14-38--32

cydia2020, 2019-09-30:
Yeah it's probably a bug the connection is quite solid

cydia2020, 2019-10-17:
Seems to be fixed in 0.6.5, closing

closed 2019-10-17 by cydia2020