commaai/openpilot #824: Is it possible to communicate with lgsvl in openpilot?

BatuhanUlucay, 2019-10-01:
If it is possible, please explain how to do that.

pd0wm, 2020-01-20:
You would have to write your own fake car. The carInterface can contain the code for interfacing with the simulator.

closed 2020-01-20 by pd0wm

lemketron, 2020-12-15:
There's a video demonstrating this (posted just before this issue was created) here: https://vimeo.com/359211939

Curious if the code has been shared anywhere or upstreamed...

vchadha, 2021-02-05:
This is very late so sorry for the necro bump, but if anyone else stumbles here, there is a really awesome paper by Sato et. al. titled "Hold Tight and Never Let Go: Security of Deep Learning based Automated Lane Centering under Physical-World Attack" where they accomplish this.

I believe they said they would opensource their bridge code between LGSVL and OP, but I am not sure if they have done this yet. Regardless, I am sure they would be open to communication.

Though OP now has a really sweet CARLA sim bridge file, so you could always use that instead.