commaai/openpilot #829: Adding comma power disables 4WH and can’t lock car with remote.

nojvek, 2019-10-06:
Tried to disable the DSU on Toyota RAV4 2016. This causes a “cruiser fault error” in Openpilot. Trying to fix that error by connecting comma AI power via obd II port causes 4WH to be disabled. Car remote isn’t recognized. You can’t lock the doors via remote. 

Is there a better way to tell the DSU to lay low and Openpilot handle longitudinal control ?

Alternatively rather than block OBD-II port, can the giraffe be powered via the cigarette lighter that USB adapters use to power the phone ?

Probably also a closer location. The OBD port to giraffe is a lot of messy wiring.

arne182, 2019-10-06:
The comma power cable can be nicely routed behind the rubber seal of the door and along the headliner. The stop switches need to be different when you use the comma power. So check that they are correct. There are also other days to power via the vanity light or the Sun roof power cable. Searching discord should give you a few answers.

ErichMoraga, 2019-10-06:
Don't forget to turn off DIP switch 3 when using Comma Power.

vanillagorillaa, 2019-10-06:
Issue here is switch 3 is high like Erich said. Closing this issue.

closed 2019-10-06 by vanillagorillaa