commaai/openpilot #831: Achieving stop and go. How to get around Low speed lockout ?

nojvek, 2019-10-06:
Really enjoying open pilot on Rav4 2016 SE. i’ve disabled DSU, set appropriate giraffe switches and see that OpenPilot can stop the car all the way to 0. But after that it asks to take manual control. Same with starting it against. Open pilot won’t kick in until speed is at 20mph. 

How do I remove this limitation? I know this could be a safety concern so OpenPilot may not take a pull upstream. I just want it for myself. 

Sitting in idle traffic is way more stressful than driving on the highway. I just want OpenPilot to go with flow of traffic. Stop when car ahead stops, go when it moves. 

I see from the code that there is a low speed lockout. Sometimes in the dbc files to. Is the lockout a hardware limitation in toyotas or this something I can disable in software and take it for a test run ?

muhatashim, 2019-10-08:
You would need to get a comma pedal in order to get a full stop and go support since there is a vehicle limitation on acceleration below 20mph. However, I actually made a commit just today on my fork that allows at least stopping and braking in stop and go traffic. It's a safety risk though on my end so don't try it unless if you really want that risk.

nojvek, 2019-10-08:
Oh I see. It’s a hardware thing then. I assumed disabling the DSU would let me get around the limit and accelerate from 0. I do see that OpenPilot stops all the way to zero and that’s working as expected. 

Pedal is cool but i’m a bit hesitant to try it out. Looking for a more software based soln.

arne182, 2019-10-08:
Give my fork a try. I am sure one of my many software solutions will help you.

nojvek, 2019-10-08:
This is your fork right: https://github.com/arne182/openpilot

Contains some cool things. Will check it out. 

It contains a lot of little features. Will need to do some diligence to ensure car won’t hold me hostage. 

Car hacking is fun but scary at the same time.

vanillagorillaa, 2019-10-10:
Seems like this question has been answered, closing

closed 2019-10-10 by vanillagorillaa