commaai/openpilot #832: Taiwan rav4 hybrid fingerprint add

maxduan123, 2019-10-06:
確定rav4 2019 hybrid台灣版可以用，可以幫我把指紋加入到dragon pilot 6.3 和 6.4中嗎? 方便我之後做教學直接推薦人們安裝dragonpilot，感恩
number of messages 93:
fingerprint 36: 8, 37: 8, 166: 8, 170: 8, 180: 8, 295: 8, 296: 8, 401: 8, 426: 6, 452: 8, 466: 8, 467: 8, 550: 8, 552: 4, 560: 7, 562: 6, 581: 5, 608: 8, 610: 8, 643: 7, 713: 8, 728: 8, 740: 5, 742: 8, 743: 8, 761: 8, 765: 8, 800: 8, 810: 2, 812: 8, 829: 2, 830: 7, 835: 8, 836: 8, 863: 8, 865: 8, 869: 7, 870: 7, 871: 2, 877: 8, 881: 8, 882: 8, 885: 8, 896: 8, 898: 8, 913: 8, 921: 8, 944: 8, 945: 8, 950: 8, 951: 8, 953: 8, 955: 8, 956: 8, 971: 7, 975: 5, 993: 8, 1002: 8, 1014: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1044: 8, 1056: 8, 1057: 8, 1059: 1, 1063: 8, 1071: 8, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1172: 8, 1235: 8, 1237: 8, 1279: 8, 1541: 8, 1552: 8, 1553: 8, 1556: 8, 1557: 8, 1568: 8, 1570: 8, 1571: 8, 1572: 8, 1592: 8, 1594: 8, 1595: 8, 1696: 8, 1745: 8, 1775: 8, 1779: 8

ErichMoraga, 2019-10-06:
@maxduan123 Please include a Cabana share link of a drive taken while using stock (1001) Giraffe DIP settings for fingerprint verification.

maxduan123, 2019-10-06:
you mean this kind of link?
https://my.comma.ai/20b4d3621d22f87b/1570352970520/1570353641114

maxduan123, 2019-10-06:
> @maxduan123 Please include a Cabana share link of a drive taken while using stock (1001) Giraffe DIP settings for fingerprint verification. Bear in mind you'll want to get your fingerprint PR'd into OpenPilot, as this is not the right place to discuss others' forks. See #custom-forks in Comma Discord, or go to the other Discord community for assistance with your other requests.

if its an drive like this, https://my.comma.ai/20b4d3621d22f87b/1570352970520/1570353641114
ill prepare one for you, if not, can you give me a hint how do I get the cabana link? thanks

maxduan123, 2019-10-06:
> @maxduan123 Please include a Cabana share link of a drive taken while using stock (1001) Giraffe DIP settings for fingerprint verification. Bear in mind you'll want to get your fingerprint PR'd into [...]

or do you mean link like this?
https://my.comma.ai/cabana/?route=20b4d3621d22f87b%7C2019-10-06--09-10-25&exp=1601933959&sig=bHaCwNZr5kmdGJyZ8Mx8W6vcm6rRWrIwBTHpA0e%2FmrU%3D&max=10&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F20b4d3621d22f87b%2F8134c89685e589069dd585c3708fdfcd_2019-10-06--09-10-25

ErichMoraga, 2019-10-06:
The 2nd one is a Cabana share link.  Hopefully you mount your EON to the windshield someday...

maxduan123, 2019-10-07:
> The 2nd one is a Cabana share link. Hopefully you mount your EON to the windshield someday...

ok, I get it here,its the taiwan rav4 hybrid 2019 driving
https://my.comma.ai/20b4d3621d22f87b/1570387424836/1570393068407

https://my.comma.ai/cabana/?route=20b4d3621d22f87b%7C2019-10-06--18-44-46&exp=1601981446&sig=SnY2iib0JjWnqlxONjitPJQqfIKEAxG7WNhHL86XAQI%3D&max=93&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F20b4d3621d22f87b%2Faab1a53e29d4c32bdcda41fc3440cc08_2019-10-06--18-44-46

please add this fingerprint into openpilot asap, so I can make a tutorial for more people in Taiwan, It should back installation in Taiwan easy and boost the number of people using it in Taiwan, thanks

maxduan123, 2019-10-07:
im using this value.py
https://github.com/dragonpilot-community/dragonpilot/blob/devel-zht/selfdrive/car/toyota/values.py
the fingerprint is
# Taiwan 2019 RAV4H from Max Duan
  {
    36: 8, 37: 8, 166: 8, 170: 8, 180: 8, 295: 8, 296: 8, 401: 8, 426: 6, 452: 8, 466: 8, 467: 8, 550: 8, 552: 4, 560: 7, 562: 6, 581: 5, 608: 8, 610: 8, 643: 7, 713: 8, 728: 8, 740: 5, 742: 8, 743: 8, 761: 8, 765: 8, 800: 8, 810: 2, 812: 8, 829: 2, 830: 7, 835: 8, 836: 8, 863: 8, 865: 8, 869: 7, 870: 7, 871: 2, 877: 8, 881: 8, 882: 8, 885: 8, 896: 8, 898: 8, 913: 8, 921: 8, 944: 8, 945: 8, 950: 8, 951: 8, 953: 8, 955: 8, 956: 8, 971: 7, 975: 5, 993: 8, 1002: 8, 1014: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1044: 8, 1056: 8, 1057: 8, 1059: 1, 1063: 8, 1071: 8, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1172: 8, 1235: 8, 1237: 8, 1279: 8, 1541: 8, 1552: 8, 1553: 8, 1556: 8, 1557: 8, 1568: 8, 1570: 8, 1571: 8, 1572: 8, 1592: 8, 1594: 8, 1595: 8, 1696: 8, 1745: 8, 1775: 8, 1779: 8
  }],

ErichMoraga, 2019-10-07:
You have this GitHub record opened as an "issue", instead of a PR (Pull Request) to change https://github.com/commaai/openpilot/blob/devel/selfdrive/car/toyota/values.py

Please close this "issue" accordingly, and open a PR.

closed 2019-10-07 by maxduan123

ErichMoraga, 2019-10-07:
Thanks.  I'll submit a PR for you later today.