commaai/openpilot #850: Add new device in support

WinKarbik, 2019-10-14:
Hello. I am start learning openpilot and as I understand for eon dashkit used oneplus 3 or leeco? Okay. I have tried compile ROM(branch mindroid) for my device (Xiaomi Redmi 3) just for testing this. Now rom compiled and started, but I see only black screen. Which application should be started after booting? I am interested in porting this for my Kia Soul Ev 2014

ErichMoraga, 2019-10-14:
The OG EON was OP3T, and the LEON (aka. EON Gold) is Le Eco Le Pro 3 (x727).  It would take a huge effort to get working on anything else.  Please join us in Discord in channels like #hw-unofficial ... as GitHub really isn't the best place to discuss this.

ErichMoraga, 2019-10-14:
Oh, and #hyundai-kia-genesis would be the right place for your 2014 Kia Soul EV, unless it doesn't even have LKAS, ACC, etc. in which case -> #old-cars

WinKarbik, 2019-10-14:
So, there is no reasons for trying this project?

WinKarbik, 2019-10-14:
And how I can find channels in Discord? I have some questions yet

ErichMoraga, 2019-10-14:
I wouldn't bother trying.  Talk to Leon, for example.  If you can't find channels, the channels will find you.  Check YouTube for Discord 101.

dorkmo, 2019-10-14:
link to discord is on www.comma.ai

closed 2019-10-15 by WinKarbik