commaai/openpilot #854: OpenPilot is still engaged after car shuts off

cydia2020, 2019-10-17:
**Describe the bug**
Prius 2018 i-Tech
OpenPilot sometimes keep itself engaged after the car is fully shutoff. OpenPilot no longer communicates with the car if the car is restarted and the car will display DSU errors.

**How to reproduce or log data**
Shut off car with OpenPilot in "ready to engage" mode

**Expected behavior**
EON should display ride completed

** Device/Version information (please complete the following information):**
 - Device: EON
 - Version: 0.6.5
 - Car make/model: Prius 2018 i-Tech

**Additional context**
Add any other context about the problem here.
Photo: 
[image]

ErichMoraga, 2019-10-17:
It sounds like you have Comma Power connected while DIP 3 is on.

cydia2020, 2019-10-19:
Negative, DIP 3 is off when it happened

ErichMoraga, 2019-10-19:
Is your DSU connected?

ErichMoraga, 2019-10-19:
It sounds like your DSU is disconnected (which is fine), although you'll be more prone to odd issues.

cydia2020, 2019-10-19:
Yes, DSU is disconnected as I don't like the factory braking model

ErichMoraga, 2019-10-19:
Your issue only happened once, or does it occur consistently?

cydia2020, 2019-10-19:
Only once, and I can't reproduce it anymore

rbiasini, 2019-10-19:
Thanks. Other users reported this issue. We are looking into it.

jpancotti, 2019-11-02:
I’ve also had this happen a few times with different versions of stock op on a 2017 CR V

rbiasini, 2019-11-02:
Should be fixed in the upcoming 0.6.6

cydia2020, 2019-11-03:
Thanks, closing

closed 2019-11-03 by cydia2020

cydia2020, 2019-11-17:
Still happened today, though the alert quickly went away