commaai/openpilot #871: 2020 Corolla Hybrid fingerprint incorrect.

midfield99, 2019-11-07:
**Describe the bug**
The current fingerprint for the Corolla Hybrid is incorrect: https://github.com/commaai/openpilot/blob/devel/selfdrive/car/toyota/values.py#L182
I ran get_fingerprint, and got this:
number of messages 118:
fingerprint 36: 8, 37: 8, 166: 8, 170: 8, 180: 8, 295: 8, 296: 8, 401: 8, 426: 6, 452: 8, 466: 8, 467: 8, 550: 8, 552: 4, 560: 7, 562: 6, 581: 5, 608: 8, 610: 8, 643: 7, 713: 8, 728: 8, 740: 5, 742: 8, 743: 8, 761: 8, 764: 8, 765: 8, 800: 8, 810: 2, 812: 8, 824: 8, 829: 2, 830: 7, 835: 8, 836: 8, 865: 8, 869: 7, 870: 7, 871: 2, 877: 8, 881: 8, 896: 8, 898: 8, 900: 6, 902: 6, 905: 8, 921: 8, 933: 8, 934: 8, 935: 8, 944: 8, 945: 8, 950: 8, 951: 8, 953: 8, 955: 8, 956: 8, 971: 7, 975: 5, 993: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1002: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1044: 8, 1056: 8, 1057: 8, 1059: 1, 1071: 8, 1076: 8, 1077: 8, 1084: 8, 1085: 8, 1086: 8, 1114: 8, 1132: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1172: 8, 1235: 8, 1279: 8, 1541: 8, 1552: 8, 1553: 8, 1556: 8, 1557: 8, 1568: 8, 1570: 8, 1571: 8, 1572: 8, 1592: 8, 1595: 8, 1649: 8, 1745: 8, 1775: 8, 1779: 8, 1786: 8, 1787: 8, 1788: 8, 1789: 8, 1904: 8, 1912: 8, 1990: 8, 1998: 8, 2015: 8, 2016: 8, 2024: 8

It also doesn't match up with a community fork.
https://github.com/markolo25/openpilot/blob/0.6.5-corollah/selfdrive/car/toyota/values.py#L193
Diff:
What the current value is missing from mine: [1076, 1077, 1084, 1085, 1086, 1132, 1649, 1164, 1165, 1166, 1167, 1786, 1787, 764,
812, 824, 1904, 2024, 1912, 900, 902, 905, 933, 934, 935, 1788, 1990, 1998, 2015, 2016, 998, 999, 10
00, 1001, 1789]
What mine is missing: [1082, 1112, 1237, 1594, 885, 918, 987, 1014]

**How to reproduce or log data**
Get the fingerprint from a 2020 Corolla Hybrid made in the US.

**Expected behavior**
I'd like a fingerprint for the 2020 Corolla Hybrid that matches up with what is on my car.

** Device/Version information (please complete the following information):**
 - Device: Not sure, I got my eon. less then three months ago.
 - Version: [e.g. 0.6.3] from an old branch from here: https://github.com/markolo25/openpilot
 - Car make/model [e.g. Toyota Prius 2016]

**Additional context**
The 2020 Corolla Hybrid only has one trim, I'd suggest renaming the current fingerprint for the Altis.

markolo25, 2020-01-28:
the reason for this is the corolla hybrid that is in the code right now is the chinese version i believe, with Fingerprinting 1 the fingerprint i have in my fork cannot co-exist with the camry hybrid fingerprint.

arne182, 2020-01-28:
Firstly. Check that your fingerprint collection is done while you at least start and stop your car 3 times to get all the signals. Then your fingerprint just needs to be a unique subset of the one in the code.

geohot, 2020-03-13:
Toyota is now exclusively fingerprinting 2.0, 1.0 will not be fixed.

closed 2020-03-13 by geohot