commaai/openpilot #877: Blind curve

arne182, 2019-11-11:
**Describe the bug**
When traveling up a hill and at the top of the hill there is a sharp curve. Open pilot has no chance to see it at the moment.

**How to reproduce or log data**
Drive on a road where there is a turn at the top of a hill

**Expected behavior**
We could use the mapd data and feed that into the model so that it knows the approximate turn that is going to come up soon.

** Device/Version information (please complete the following information):**
 - Device: both
 - Version: 0.6.6
 - Car make/model: All

geohot, 2020-01-23:
This will be improved with a larger model box. No plans to revive mapd, the underlying data source (OSM) is not high quality enough in the USA.

closed 2020-01-23 by geohot

arne182, 2020-01-23:
How will a larger model box know that above the hill there is a sharp corner without map data?