commaai/openpilot #881: Toyota dash speed not matching openpilot

arne182, 2019-11-17:
**Describe the bug**
The speed indicated on the Toyota dash is not the same as on the open pilot display. I assume that it is not planned to match those speeds as it would not correspond correctly to the obd speed.

The main reason why I am not more engaged on stock is because the minimum set speed is 41 kph. Running my fork gets you 7kph minimum set speed and that allows for a whole new range of low speed capabilities. 

**How to reproduce or log data**
enable open pilot and see the offset and try make a 90° turn.

**Expected behavior**
If below 41kph engage don't accelerate to 41kph but keep the current speed as a set speed.

** Device/Version information (please complete the following information):**
 - Device: EON/EON Gold
 - Version: 0.6.6
 - Car make/model: Toyota/All

**Additional context**
If you are anyway not tracking the correct indicated speed then we can completely change it. I have also enabled now 5 kph when double tapping the speed stock or long holding it. That makes for an interesting drive as you can control the set speed much faster.
Running at a speed lower than the set speed is fine but running at a high speed causes the car to fight against the acceleration and so the speed increase can only happen slowly in the beginning.

cydia2020, 2019-11-21:
That's Toyota's problem, they purposely increased the speed displayed by a small margin to please their lawyers. The vehicle however, knows exactly how fast it's travelling, it just doesn't wanna tell us.

arne182, 2019-11-21:
I think I need to put a PR in place for this. Just have 1 last bug to work out.

geohot, 2020-03-27:
openpilot should always show the true speed of the vehicle. Soon, this will go through locationd.

closed 2020-03-27 by geohot