commaai/openpilot #883: Waiting Toyota dash display

arne182, 2019-11-19:
[image]

**Describe the bug**
If you have ever seen this dispaly before above. It is because you have probably a Toyota hybrid and you have come to a complete stop and now you can not use the resume button anymore to up the speed

**How to reproduce or log data**
Grab a rav4 hybrid. Take out the dsu and try stop and go.

**Expected behavior**
A clear and concise description of what you expected to happen.

** Device/Version information (please complete the following information):**
 - Device: EON/EON Gold
 - Version: 0.6.6
 - Car make/model: Toyota Hybrid 2017 maybe more Hybrids effected

**Additional context**
This state arises because of disengage on gas. If you would be able to tap the gas pedal after a stand still you would be allowed to up the set speed by using the stock. 

A general safety question would be is it safe to disengage lateral control on gas or brake? Should we not consider using the same approach as lateral for longitudinal. You can help open pilot along by giving more gas when needed or more brake when needed without disengaing. It would be crazy if with every lateral input you would loose longitudinal control and so I would propose for a change in the safety rules.

ErichMoraga, 2019-11-19:
I believe what you're talking about is Comma's all or nothing approach, which I'm fine with.  Of course users will always want hackery, and throw caution to the wind.

zorrobyte, 2019-11-22:
I second @ErichMoraga 's notion of OP having well-defined states. In an emergency situation or someone new driving your car; the ability for OP to disengage on gas or brake is an important safety feature, imo.

Being able to tap gas at a stop to resume cruise could be a cool feature to have for this narrow use case (from a stop, DSU initiated standstill, tap to resume), but I'm not personally going to seek this enhancement until OP long control is usable (I drive with DSU connected).

pd0wm, 2020-01-20:
This seems to be a "bug" in the Toyota Hybrid since it's not designed to handle stop&go. Not much we can do from the openpilot side.

We won't allow being engaged after a gas press.

closed 2020-01-20 by pd0wm

arne182, 2020-01-24:
Understandable, the issue is not after the gas press. The solution is after the gas press. So hybrids that stop will not be able to increase set speed until they reengage. On my fork I at least give the users an alert that this is the case. Or do you not want to alert the user?

iamjackg, 2022-06-01:
Should this go away now that 8.14 will allow OP to stay engaged on gas? I.e. I'll just have to tap the gas after a standstill to restore normal behaviour?