commaai/openpilot #886: Toyota Giraffe Configuration Fix - Medium - Not working with new Toyota Giraffe

onekrishnakumar, 2019-11-23:
**Describe the bug**
I got the Unsupported Giraffe Configuration message. I went to comma.ai/tg and was directed to https://medium.com/@comma_ai/toyota-giraffe-configuration-fix-20283ba4d01a
image.png
Based on the recommendation in the article, I bought a new Toyota Giraffe, which comes with a built-in 120 Ohm resistor. And then I simply pulled the giraffe switch 2 to high. I also made sure that the giraffe is plugged into the stock camera. Then, I restarted my car and checked if the alert on my EON display disappeared. And it Did!
Then I went back to the original setting on the giraffe switch 1,2,3 to high and 4 to low. But when I try and engage open pilot. It does not engage and an error message displays. The error message says, communication error between systems. Then the error message disappears but open pilot never engages. :(

**How to reproduce or log data**
https://my.comma.ai/172a1afdf159d8ff/1574486729234/1574487060007

**Expected behavior**
The open pilot should have engaged.

** Device/Version information (please complete the following information):**
EON
 0.6.5
Toyota Prius Prime 2018

**Additional context**
Would appreciate any help

ErichMoraga, 2019-11-23:
Why are you using 1110?  Go back to 0101 w/ Comma Power, as you clearly observed that worked.  No need to open a GitHub issue when this is discussed quite often in Discord.

Also, feel free to read the Giraffe gotchas in the release notes...
https://github.com/commaai/openpilot/blob/devel/RELEASES.md

ErichMoraga, 2019-11-23:
Giraffe shipped w/ 0101 pre-configured, and that's supported (use it w/ Comma Power).  No need to open a GitHub issue when this is discussed quite often in Discord.

Also, feel free to read the Giraffe gotchas in the release notes...
https://github.com/commaai/openpilot/blob/devel/RELEASES.md

onekrishnakumar, 2019-11-23:
Hi Erich,

I was not using it with Comma Power, that is why it was not working!
Thank you for stating that. Now it is working with Comma Power!
Even the Medium article did not say to use it with Comma Power!
Now it is working with Comma Power!

Thank You,
Krishna.

closed 2019-11-23 by onekrishnakumar