commaai/openpilot #899: how to calc gradient descent in file params_learner.cc

gucasbrg, 2019-12-03:
I can get the formula from vehicle model,
[image]
but how to calc gradient descent angle offset, tire stiffness and steer ratio from this formula?

vangogh0318, 2019-12-03:
I've been looking at  params_learner.cc  file recently. Can you explain the formula in it? thanks

pd0wm, 2019-12-03:
```python
from sympy import *

sa, u = symbols('sa, u')
ao, x = symbols('ao, x')

# vehicle params
sR, cF0, cR0, aR, aF, l, m = symbols('sR, cF0, cR0, aR, aF, l, m')
cF = cF0 * x
cR = cR0 * x

psi = symbols('psi')
chi = 0.

sf = m * (cF * aF - cR * aR) / (l**2 * cF * cR)  # Slip factor
curvature_factor = (1. - chi) / (1 - sf * u**2) / l
curvature = curvature_factor * (sa - ao) / sR
yaw_rate = curvature * u

print("ao", simplify(diff((yaw_rate - psi)**2, ao)))
print("x" , simplify(diff((yaw_rate - psi)**2, x)))
print("sR", simplify(diff((yaw_rate - psi)**2, sR)))
```

closed 2019-12-04 by gucasbrg