commaai/openpilot #910: Openpilot 2019 camry

natejc39, 2019-12-12:
**Describe the bug**

 I saw that your car is being "fingerprinted" (the process by which openpilot determines what car it is running on) as a 2019 RAV4, thus the question. Since your car is not a 2019 RAV4 is is being "mis-fingerprinted" and not properly identified.

**How to reproduce or log data**
Steps to reproduce the behavior, or a explorer/cabana link to the exact drive and timestamp of when the bug occurred.

**Expected behavior**
I’m hoping someone can guide me on how to do this?  Thank you in advance. 

** Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold]
 - Version: [e.g. 0.6.6],
 - Car make/model 2019 Toyota Camry SE

**Additional context**
Add any other context about the problem here.

arne182, 2019-12-12:
Can you give me your fingerprint with the script /data/openpilot/selfdrive/debug/get_fingerprint.py

ningfeiwang, 2020-02-06:
Hello,
I got the same problem as him which is my 2019 Camry SE recognize my car as RAV4 2019 by OpenPilot 0.7.1. Is there any solution for it?

pd0wm, 2020-03-18:
Is this still the case with ECU FW based fingerprinting? Can you share your dongle id if that's the case?

closed 2020-05-11 by pd0wm