commaai/openpilot #913: Learned angle offset not accurate

ShaneSmiskol, 2019-12-14:
**Describe the bug**
The average angle offset that openpilot dumps into LiveParameters never seems to be accurate, which I think has been the cause of my lane hugging issues since I got openpilot about a year ago. Since my car had 50,000 miles put on it in the course of the year prior to me buying it, I assume it developed some wheel angle offset, or misalignment with the tires causing all this. I first tried messing with the camera offset, then different mounting positions and angles only for nothing to help my left lane hugging.
My `angleOffsetAverage` value right now is `-0.080168674153025768`, and it never seems to stray too far from that value. However when on Arne's fork a few months ago, I noticed that when I or openpilot is driving on a straight highway, my steering wheel is at around -1 to -2 degrees (pointed to the right). I even notice myself naturally offsetting the wheel to the right slightly in order to go straight. After adding ~1.5 degrees in `latcontrol_pid` to my current angle_steers (so that -1.5 is 0 degrees), my left lane hugging around left curves is nearly non-existent.

**Device/Version information (please complete the following information):**
 - Device: EON Gold
 - Version: 0.6.6
 - Car make/model Toyota Corolla 2017 LE

ErichMoraga, 2019-12-15:
And have you taken it to an alignment shop yet?

arne182, 2019-12-15:
Doing the wheel alignment sounds like a good idea. The issue that I find is that when they then calibrate the sensor off data going straight for a few hundred meters it is often off by about a few tenths of a degree, sometimes even up to 2°.

ErichMoraga, 2019-12-15:
When Shane said, "...notice myself naturally offsetting the wheel to the right slightly in order to go straight"... it pointed to an actual hardware (chassis, tire, stiff wind, etc.) or alignment issue w/ his car.  Best way to fight a hardware problem is w/ a hardware fix.

ShaneSmiskol, 2019-12-18:
@ErichMoraga The problem that would require a hardware fix, time, and money can be mostly fixed by manually changing my steering offset in latcontrol. Lane centering, while still not perfect, has been better than anything I've tried in the last year I've had openpilot. I'm just curious why it's failing to correctly detect any significant offset for my car

ErichMoraga, 2019-12-18:
You're not concerned about improper tire wear?  Additionally, you're more likely to lose control of the vehicle when there's a traction issue (eg. heavy rain).  Improper alignment can really be a safety issue if you let it get out of hand.

closed 2019-12-30 by ShaneSmiskol