commaai/openpilot #916: Slow cruise control breaking, not using engine breakingg

spairrow326, 2019-12-15:
**Describe the enhancement**
Enhancement: Cruise control breaking slow to respond to lowering set speed and when going downhill openpilot only uses breaks not enough downshifting engine breaking. Stock cruise control breaking and downshift engine breaking seems better tuned then openpilot. 
**How to reproduce or log data**
Change set speed from 50mph engaged to 19mph engaged as fast as possible VS. 50mph to 19mph with stock 2020 corolla le, Stock cruise control breaking more fast and responsive. When going down a long hill stock system downshifts but openpilot only uses breaks but no engine breaking.

**Expected behavior**
More responsive cruise control breaking when changing set speed. More downshifting breaking when going downhill.

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold]
 - Version: [e.g. 0.6.6], 
 - Car make/model [e.g. Toyota Corolla LE 2020]

**Additional context**
When openpilot cruise control breaking is slow to respond to the changes in set speed it is less safe and i can't keep the system engaged as long as i would like because i have to hit the breaks. When openpilot engaged in long downhill it wears down the break pads but stock system uses more engine breaking.

ErichMoraga, 2019-12-15:
Sell breaking > buy braking

cydia2020, 2019-12-15:
Yes noticed it as well when going through a long descent, normally if I press the brake pedal the battery charges a lot faster than using openpilot on the Prius.

arne182, 2019-12-15:
On my RAV4 I have exactly the opposite. Braking is so hard that the regen braking kicks in. but when accelerating openpilot puts so much gas that it knocks it out of EV mode making the engine switch on every time it accelerates.

cydia2020, 2019-12-15:
Yes that's why I don't usually accelerate with openpilot, and switch to B gear just after I switch off openpilot to bring the car to a stop, otherwise it'd be way too inefficient

arne182, 2019-12-16:
https://github.com/arne182/openpilot/commit/485b84caa787b093c8109a3f8a21ebcca8c5bcf2

This might do what you want. I have linked the deceleration profiles to the eco and sport buttons. Happy braking. Maybe something for comma to consider.

nelsonjchen, 2020-04-10:
I've been experimenting for quite a bit with this for the past month on and off. It's such a pain to test this in Phoenix. I have to drive out quite a ways to get usable hills and I'm a little jealous of San Diego and its freeway or wide road hills on demand in the canyons at times. It's nice to have a reason to go out though in these times. 

So in my 2020 Corolla Hatchback, I observed some interesting behavior.  

Apparently, you know that difference in the dash cluster? Like if you tell OP to set a speed of 45 MPH and but in the dash cluster you see 47MPH as the set cruise speed? That's actually might be related. 

On an empty, remote, and gently downward sloping road in the desert at night that I've confirmed would engine brake with the stock Toyota longitudinal, I temporarily tried and checked out a branch I've prepared that that disables sending acceleration commands below 0.0 when there's no lead. *Just to be clear, this was by no means a build meant to be usable on normal roads*. Hands on the wheel, foot almost caressing the brake, with the modified test OP cruise set to 45MPH, 47MPH on the car dashboard, I observed that once the car reached 47MPH, the car would downshift and actually engine brake. 

I should add that I also recreated during that test the what I assumed to be "permit braking" (`PMTBRKG`?  Assumed from leaked DBC https://github.com/commaai/opendbc/blob/36c471e59eaac3760a00125d557ef19af091f289/toyota_2017_ref_pt.dbc#L104) bit and an additional copy of the acceleration bits that stock Toyota longitudinal also sends but OP doesn't (`AT_RAW` in the leaked DBC as well; I don't know what that might stand for and is `AT` Automatic Transmission or something else?). `PMTBRKG` or "permit braking" goes high when a lead car is detected. I haven't done more experiments yet to see if these are purely informational or not with regards to engine braking. I think "permit braking" is informational and `AT_RAW` or `ALT_ACCEL_CMD` as I've called it in my changes is not informational but I'll need to do experiments to check.

What I do think might be happening is that the ECU for the transmission is also listening to the cruise control setpoint. It appears on stock Toyota long, setting a cruise speed of 45mph in the cluster would really go 43mph but if the car really actually reaches 45mph due to additional acceleration from stuff like the downhill, it'll engine brake. 

As a proof of concept, I prepared a branch where the planner and UI displaying the target speed was modified to multiply the target velocity by some factor to roughly mimic the car's dashboard. Multiplying the `v_cruise_setpoint` in the planner and `s->scene.v_cruise` in the UI painter by `1.02` along with my extra DBC bits recreation of `PMTBRKG` and `AT_RAW` and it seems to give engine braking behavior that's much closer to stock while still keeping OP in control of the longitudinal including curves and stuff at OP's determined speeds. Also, if that factor of `1.02` is too high, on flat ground, the car will downshift, and run the engine at a too high of an RPM.

It just might be that Toyota has a chill roughly 2 mph+ window at around 45 mph in their long instead of OP's more less chill and precise window that doesn't go above the dashboard displayed setpoint and Toyota's window at the car's setpoint. Perhaps the engine braking is Toyota's way of slowing the car down without causing the taillights to go on when it is about to exceed a "speed limit setpoint" value on the dashboard? 

I i'm not sure if the proof of concept approach might be right though. This might be the wrong place to add that factor as well since its some of OP's internal processing. It might be a better idea to put that into the generated Toyota DBC to scale the setpoint from the car and have the whole rest of OP be none-the-wiser.  

Also, there's still some jank sometimes in corners where it may add gas coming out of them; I'm not sure if Toyota long recognizes that its engine braking/coasting downhill and lets the downhill acceleration restore the speed or if it's the current OP long's infamous aggressiveness. Another jank is that the transmission ratio for engine braking might not be perfect. It sometimes downshifts too much, slows too much, and adds a bit of gas on the downhill which is a ridiculous event to see in my CVT Corolla with its ability to theoretically have a perfect ratio. An imperfect transmission ratio is a behavior of which I can't say I've seen from Toyota's long.

HaraldSchafer, 2021-12-15:
Cruise control braking is a lot better now in openpilot than when this issue was opened.

As for engine braking, it doesn't seem like that's something we can reliably control. The way toyotas engine brake doesn't seem like a desirable way to control the engine braking, especially since it seems different even between toyotas.

closed 2021-12-15 by HaraldSchafer