commaai/openpilot #921: Forward Collision Warming

xuchongang, 2019-12-17:
When I upgraded to 0.7 stock,  with Honda Odyssey, there are way too many false positive warning on two-lane roads. It happens almost 4-5 times on a 30 minutes drive. Would appreciate if comma could investigate this as it could be dangerous if the brake is applied. So far, none of these cases applied brakes yet, but I see the security issue if it is applied. The DBC files were updated for 0.7 and is possible it is linked to that?

rbiasini, 2019-12-17:
@xuchongang, links would be great, thanks.

xuchongang, 2019-12-17:
@rbiasini , will the HUD record the screen of FCW? Seems that I cannot find the specific segmant with red screens.

rbiasini, 2019-12-17:
Just the link to the drives are fine. Thanks

xuchongang, 2019-12-17:
Please see the link below: https://my.comma.ai/6484765a32a70156/1576547083156/1576547945730

pd0wm, 2020-01-20:
The Honda Nidec stock system seems to have many false positives.

It also starts with just visible warnings, and starts beeping later. When you made this issue we triggered an audible warning from the EON when the Honda system just did a visible one. This has since been fixed.

I'm closing this issue now, but feel free to open a new one if you still feel like you get more warnings then with just the stock system running.

closed 2020-01-20 by pd0wm