commaai/openpilot #926: Improving the vehicle control using machine learning

xuchongang, 2019-12-18:
Based on my own experience and conversation with many users of openpilot throught discord,  tuning might be needed to get a good and stable drive for different models.  This really affected the user experience. Furthermore, the learned steering ratio might change through time and thus different tuning might needed. My exploration of the openpilot codes showed that openpilot uses 3 controllers: PID, INDI and LQR. The PID was used in most of the models and has the general issue of wobbling steering if it was not well tuned.  For the future improvement, it would be great if comma can use machine learning to apply dynamic torques along the road just as human drivers did. I searched online, it seems that there are already some prelinary studies showed the promising of that. For example, https://arxiv.org/pdf/1810.09365.pdf.  

The openpilot is great and if comma can make this part better, it would substantially increase the  user experience. In any case, I really hope that comma can make improvement in this area. Thanks again for your great work from the comma team.

ErichMoraga, 2019-12-18:
Not sure what car you drive, but on the Prime, I spent month messing w/ multi-breakpoint PID tuning after I installed OP in 2017.  OP's PID implementation is partially incomplete, though.

Anyway, after installing ZSS (high resolution steer angle sensor), I've found that basically eliminated ping pong.  My branches use that w/ INDI for super smooth drives.

Be sure to check your Cabana logs, and inspect the resolution of your sensor accordingly.

xuchongang, 2019-12-19:
@ErichMoraga, thanks for the comments and seems INDI is better than PID. Did you check the run without ZSS but with INDI? Does it improve the performance compared to PID? If it is, it might be good that comma move more models to INDI.

ErichMoraga, 2019-12-19:
I still get ping pong if I only run INDI without the hi-res sensor.

xuchongang, 2019-12-19:
In this case, your experience indicates that the problem is the steering angle sensor. Namely, the car does not provide good steering angle data to OP and thus cause inaccurate control on the steering.

ErichMoraga, 2019-12-19:
The sensor is consistent with other TSS-P Toyotas (fast updates, but low resolution of ~1.5 degrees), which is why I asked what you drive, and if you examined your sensor data.  Since the ZSS is hi-res, it can differentiate between 0 degrees and .5 degrees, whereas the stock sensor would see no difference, and basically act like it was unpredictability out of sync with reality.   The ZSS effect is no ping ping in straights or curves.  Tuning is a thing of the past.  I have pics pinned in #toyota-lexus

xuchongang, 2019-12-19:
This is really useful information. I drove Honda Odyssey and I do not think there is a ZSS sensor available for the car. Is it difficult to install ZSS for your car? I did not know how to check on the angle resolution. I paste a shared link here if you could help. https://my.comma.ai/cabana/?route=6484765a32a70156%7C2019-12-18--16-52-56&exp=1608252496&sig=tdIixdrYrQG62SRZVgT8CaQ1DTg6LI%2Bx%2BdWJM24I7Bc%3D&max=-1&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F6484765a32a70156%2F098c95a691eaaf449d47a2e55ddb7e30_2019-12-18--16-52-56

ErichMoraga, 2019-12-19:
I have videos on my YouTube channel showing how to install it.  High-level, it's really just a matter of attaching a tiny, cylindrical magnet to the steering/EPS shaft, and positioning the sensor board so the magnet is centered over the sensor chip.

As for your Cabana data, the video loads, but it's stuck loading messages from your Odyssey.  Determining the sensor's resolution is simply a matter of looking at the "steps" in the plotted data.  When a wheel is turned, the data should appear smooth, not stepped.  The distance between each step is the resolution.

xuchongang, 2019-12-19:
Thanks. Can you share the youtube link. How about this link? https://my.comma.ai/cabana/?route=6484765a32a70156%7C2019-12-17--17-14-11&exp=1608254053&sig=wVqqjftQQgKl6tPZVrZdwE6AwXUNU8SsA4L2lSVUfD8%3D&max=6&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F6484765a32a70156%2Fcabbdf735e171dba6ea524ae9f51cd64_2019-12-17--17-14-11

ErichMoraga, 2019-12-19:
Video 1 of 10 (quick clips) showing ZSS installation...
https://www.youtube.com/watch?v=XS66gC_mQtE

Anyway, your sensor's resolution appears to be .1 degrees (not too bad)...
[image]

xuchongang, 2019-12-19:
Thank you very much for all the information!

xuchongang, 2019-12-19:
@ErichMoraga do you need special version of op to take advantage of the higher resolution angle sensors?

ErichMoraga, 2019-12-19:
Extremely minor modifications, as seen in my fork.

closed 2020-06-23 by adeebshihadeh

Samcat308, 2022-06-13:
I have a 2017 Prius prime with a new install comma3 running stock openpilot can I please get a link to purchase ZSS I've looked and had no success on Discord 
Thank you for your time

ErichMoraga, 2022-06-13:
#for-sale channel.

Samcat308, 2022-06-13:
I'm sorry I'm not familiar with how to navigate to merchandise how do I purchase ZSS for 2017 Prius prime

Samcat308, 2022-06-13:
I I.M. you hoping to get a link I'm sorry I not seeing it on GitHub

ErichMoraga, 2022-06-13:
https://discord.comma.ai/

Samcat308, 2022-06-13:
Hello Erich
I’ve been trying to navigate on Discord with no success hoping I can purchase a ZSS is it possible to get one from you?
I have a 2017 Prius prime using a comma3 stock open pilot.
Sent from All-Ways Painting 

> On Jun 13, 2022, at 12:29 PM, Erich Moraga ***@***.***> wrote:
> 
> #for-sale channel.