commaai/openpilot #930: Pedal tuning (17 Corolla)

ShaneSmiskol, 2019-12-27:
I took a drive on the stock Toyota pedal tuning (albeit not on stock openpilot, but there were no leads so it was just maintaining set speed) and this is how my velocity and set speed relate:

[image]

The time that set speed was set to 70+, I was driving on open road without any major hills. The maximum it overshot was 3 or 4 mph at 75 mph, while it was mostly 1 or 2 mph overshoot. It overshoots even if it's a slow acceleration, it continues to give gas when above by around 1.5 mph then it figures out that it needs to slow down after a second or two.

I really have no experience tuning PID or PI loops, so I have no real solution. But if the raw data can be useful to anyone who's more experienced, let me know.

ShaneSmiskol, 2019-12-30:
[image]
This was my acceleration command from openpilot (/the car?) For some reason, it's always requesting ~.11m/s/s until it realizes that the speed is higher than my set speed and requests brake. Does anyone have any clue why this is?

arne182, 2019-12-30:
I can account for 0.02 for hysteresis
ACCEL_HYST_GAP = 0.02  # don't change accel command for small oscilalitons within this value

And 0.06 because of the abs pump fix.
if CS.CP.enableGasInterceptor:
      # send only negative accel if interceptor is detected. otherwise, send the regular value
      # +0.06 offset to reduce ABS pump usage when OP is engaged
      apply_accel = 0.06 - actuators.brake
apply_accel, self.accel_steady = accel_hysteresis(apply_accel, self.accel_steady, enabled)

Where the rest of the 0.03 is hiding I am not sure.

ShaneSmiskol, 2019-12-30:
I took a short drive at 50 mph with 0 integral and I was pleasantly surprised that it didn't overshoot, it actually let off the gas as I approached my set speed when with integral (above) you can see it overshooting

arne182, 2019-12-30:
For the math down.
0.06% gas for abs pump. Minus 0.02% gas for hysteresis. 0.04% times 3.000 because it is the brake maximum value and the scale value is based on the maximum of both brake and gas maximum. Which gets you to 0.12 apply steer, give or take a few rounding errors

pd0wm, 2020-01-20:
We should probably have different pedal tuning for different cars.

Once you have the tuning figured out, I'd love to merge in a PR if someone else has tried it too.

ShaneSmiskol, 2020-01-20:
@pd0wm Want to know what you make of this: I just took a drive using stock longitudinal tuning (for non-pedal Toyotas) and it was almost perfect, much better than the pedal tune. At the crest of a long hill, it only overshot by 1 mph at 75 mph. That's the worse situation that I could have been in on the pedal tune, it would overshoot by at least 3 mph since it builds up integral going up hills like that if proportional isn't enough (I'm guessing).

I don't have raw data yet, but I'll gather some soon and plot the difference on a graph.

pd0wm, 2020-01-21:
Overshoot sounds like too much integral.

Another issue could be a non-linearity in the pedal. I.e. 50% pedal action causes different accelerations based on the speed. The PID controller assumes that a certain amount of pedal action always causes the same amount of acceleration.

For the Acura pedal we solved this by training a small neural network to estimate the required pedal output for a given desired acceleration and speed: https://github.com/commaai/openpilot/blob/master/selfdrive/car/honda/interface.py#L57

You don't have to do this with a neural network. A simple function that increases pedal action at higher speeds might already do the trick. Try something like:

```python
def compute_gb(accel, speed):
  return float(accel / 3.0 * (0.5 + speed / 60.0)) # 0.5x at 0mph, 1x at 70mph
```

ShaneSmiskol, 2020-01-21:
I have a bunch of driving data from a few different makes and models and some experience with Keras so I'll definitely try to see if I can come up with a model that closely matches the one in the Honda interface file so I can just paste in the weights! Thanks for that, I didn't know a model was being used to get a better response in the Acura and it never occurred to me that it might not be possible to fix this behavior by changing the tuning values alone.

So despite the input variable being named `accel`, it's not desired acceleration, right? It's just the PID's output that may or may not closely resembles desired accel?

ShaneSmiskol, 2020-01-22:
I tried a very similar compute_gb function to the one you posted here and it doesn't seem to help overshoot, it's still winding up and giving gas up to 3 mph above set speed at 70mph. Increasing integral somehow nearly fixes it though, while on a RAV4 (kumar's tested it), it doesn't give as good results as my Corolla.

@pd0wm Would you merge a PR if I changed just my Corolla to always use stock tuning (not pedal)?

ShaneSmiskol, 2020-01-23:
I tried a function very similar to your compute_gb above and it had little or no effect, it was still winding up and overshooting at high speeds. The non-pedal tune with my pedal and some logic to reset integral when the PID error crosses or is zero works nearly perfectly though. Without the logic to reset integral, the higher long PID tune still overshoots about the same as before

pd0wm, 2020-01-23:
Resetting the integral on zero error is not a feasible strategy. That is asking for oscillations. Integral solves error -> Integral is reset -> controller output lowers -> error increases -> integral increases again.

ShaneSmiskol, 2020-01-23:
I know what you mean, it causes oscillation for other people who have tried it, but it works perfectly for me, despite the fact that it *should* oscillate. I don't understand it, but it keeps me at my set speed perfectly.

ShaneSmiskol, 2020-03-27:
I have a few updates to this issue; I tried adding a derivative term to the PI loop for long control however it didn't work too well since the setpoint changes *with* the measurement (our speed). However if I limited derivative to only be active when the setpoint isn't changing much (we're cruising at x speed without any lead) it actually ended up working much better.

I was able to improve this performance even better when I instead added derivative to integral, it winds down integral as we approach the set speed which performed hugely better when I went over hills, or engaged from a stop and it accelerated to something high like 50 mph.

ShaneSmiskol, 2020-03-28:
Here's a graph I've made up of a short ~3 minute-ish drive I took (circled back and took the same section of road).

[image]

I luckily hit the same traffic light after decreasing my speed to 53 mph so I could run that 0 to 50 mph test twice.

You can see that with derivative the average error is decreased by almost half each time. Still some tuning to do, it's not perfect yet, but much better.

ShaneSmiskol, 2020-03-28:
Here's a section of road that I was able to drive uninterrupted by any leads for a longer time than above:

[image]

However on the way back traffic picked up so I wasn't able to drive the same route again without any (slower) cars ahead of me. You can see the average error is even better at 0.09 m/s. I will try to do that section of road again without derivative tonight and post here with a comparison

xps-genesis, 2020-05-13:
> Overshoot sounds like too much integral.
>
> Another issue could be a non-linearity in the pedal. I.e. 50% pedal action causes different accelerations based on the speed. The PID controller assumes [...]

Does it make more sense to close loop with engine torque output? Do we this info on the CAN? Engine Torque at the axle to vehicle acceleration math is straight forward, that way at any speed a given engine torque will produce the calculated acceleration in the vehicle?

ShaneSmiskol, 2021-06-03:
Issue is that we send acceleration to the pedal when it expects gas. Above 19 mph is fixed here: https://github.com/commaai/openpilot/pull/21101

closed 2021-06-17 by adeebshihadeh