commaai/openpilot #938: Add fingerprint for car using Maestro RR

dcantatore, 2020-01-05:
Hi,

I was wondering if it would be reasonable for me to submit a PR adding in the fingerprint for Prius' using a Maestro RR. The Maestro RR is device used when replacing your stock stereo system to maintain functionality from your steering wheel controls and a few other things. 

Based on the Prius Prime having a awful entune stereo system I think more and more people will be doing this as time progresses. Also, I have to add the fingerprint manually every time I update versions automatically.

ErichMoraga, 2020-01-05:
As time progresses, everyone will be using VIN Printing.  Your configuration is extremely uncommon.  For ZSS (which also also adds a msg.), I host my own branch, for example.  You could do the same for now.

pd0wm, 2020-01-17:
As @ErichMoraga says this should no longer be an issue after 0.7.2 when we switch to firmware based fingerprinting.

closed 2020-01-17 by pd0wm