commaai/openpilot #941: Comma EON / Android app as standalone for generic cars

cristipurdel, 2020-01-06:
This is more of a feature than a bug report.
It would be nice to buy the Comma EON or to install the software as an Android app in order to use it for cars which are not currently supported.
The Comma or app should just audio warn the driver in case of FCW, LDW or driver sleepiness but not interact with the car. Later on warning features like red light detection/ speed limit could be added even interacting with the OBD to break in case of FCW.
In summary it would be nice to backport at least the warning system to cars which are not supported but would benefit from the logged miles.

ErichMoraga, 2020-01-06:
Use #feature-requests on Discord.

geohot, 2020-01-23:
No plans to do this. Dedicated hardware is the path forward.

closed 2020-01-23 by geohot