commaai/openpilot #951: Peugeot 3008

cristianku, 2020-01-17:
Hi, do you plan to have it compatible with Peugeot 3008 ?

ErichMoraga, 2020-01-17:
Some would have to port it.  If it has LKAS, you can certainly attempt it yourself.

cristianku, 2020-01-17:
I can help port it. I am a developer. but what about LKAS, how can I find out about this LKAS ?

cristianku, 2020-01-17:
I mean it has a form on lane keep assist that does not keep in the lane but it correct the steering to bring back into the middle of the lane if I am near to surpass it

cristianku, 2020-01-17:
And the porting require some reverse engineering on the car protocol to understand how interact with the electric steering wheel or what ?

bugsy924, 2020-01-17:
Send me a message on discord.comma.ai (Bugsy) and I'll show you the ropes assuming your camera and EPS communicate over a standard CAN bus. If the assumption is true, then you should be able to get it steering in a few days after receiving the hardware

geohot, 2020-01-23:
Closing issues requesting car ports. Pull requests with car ports in them will be reviewed!

Get started on discord: discord.comma.ai

closed 2020-01-23 by geohot

jcmichot, 2022-11-01:
any feedback about Peugeot 3800/5800 ?

dv336699, 2023-04-05:
> I would be very much interested in this. I have a 2017 5008 and it seems to me it is equipped with everything needed for a module like this. Unfortunately no support yet. Is it planned?

I don't think commaai is going to implement it by themselves, so it's down to the community to do it I'm afraid