commaai/openpilot #952: 0.7.1 panda subaru safety has a logic bug which causes eyesight to fault on engagement

martinl, 2020-01-17:
**Describe the bug**
Subaru RX check in panda safety includes both preglobal and global platform can messages which causes eyesight to fault when openpilot is engaged

https://github.com/commaai/openpilot/blob/devel/panda/board/safety/safety_subaru.h#L14
```
AddrCheckStruct subaru_rx_checks[] = {
  {.addr = {0x119, 0x371}, .bus = 0, .expected_timestep = 20000U},
  {.addr = {0x240, 0x144}, .bus = 0, .expected_timestep = 50000U},
};
```
Global platform models have only 0x119 and 0x240 and preglobal platform models have 0x371 and 0x144. I think Subaru RX check fails because it requires that all 4 exist.

For a temporary workaround, I disabled addr_check and addr_check_len in subaru_hooks and after reflashing panda eyesight did not fault when openpilot is engaged:
https://github.com/commaai/openpilot/blob/devel/panda/board/safety/safety_subaru.h#L154

**How to reproduce or log data**
Engage openpilot using Eyesight SET button

**Expected behavior**
Eyesight should not fault when openpilot is engaged

**Device/Version information (please complete the following information):**
 - Device: EON
 - Version: 0.7.1
 - Car make/model: Subaru Crosstrek 2018

**Additional context**
Add any other context about the problem here.

rbiasini, 2020-01-17:
At least in its intent, It shouldn't cause a problem: when 2 addresses are specified in the same check, it means that either of the 2 messages are enough to pass the check.

martinl, 2020-01-17:
bus 0 is correct for both 0x119 and 0x240. I'll do some more testing with only global messages and higher timeout values

rbiasini, 2020-01-17:
oh, I thought the Subaru Crosstrek 2018 was non-global.
Ok, thanks for looking into this.

pd0wm, 2020-01-18:
Should be fixed by: https://github.com/commaai/panda/commit/f67ec28e3fd8b3239542bada8259982061b89442

Also pushed that change to devel. Can let me know if your problem is solved?

martinl, 2020-01-19:
Thank you, openpilot engages without issues after pulling hotfix and reflashing panda fw. Did about 30min city/highway drive for testing with Crosstrek 2018

pd0wm, 2020-01-19:
Thanks for testing!

closed 2020-01-19 by pd0wm