commaai/openpilot #969: UI does not draw leadTwo

pd0wm, 2020-01-18:
When a cut-in happens (which will be present in `radarState.leadTwo`) this is not drawn on the UI. The car will start braking before the lead car shows up in the ui.

pd0wm, 2020-01-20:
When there is no cut-in leadOne ad leadTwo have very similar or exactly the same values. When any of the position variables dRel/yRel diverges more than 0.1 meter they can be considered different.

AravindaDP, 2020-02-19:
Does this issue still open to be picked up?

Like to contribute doing this. Per my understanding;
Current behavior: Planner already accounts for both leadOne and leadTwo but UI only draws leadOne
Expected change: Change appropriate code in selfdrive/ui (paint.cc, ui.c and ui.hpp I suppose)

I see that lead will be accounted for both drawing chevron and track height. Does track height should be based on nearest lead in this case?

Thanks in advance

arne182, 2020-02-20:
https://github.com/commaai/openpilot/pull/1133

Done!

[image]
Tested!
Working!

closed 2020-02-20 by pd0wm