commaai/openpilot #978: Missing car models in tests

quillford, 2020-01-20:
There's an extensive [list](https://github.com/commaai/openpilot/blob/79122ae66230cac5457e09011435373967d0e1bf/selfdrive/test/test_car_models.py#L414) of models that aren't tested in this repo. When this list is complete, CI should [fail](https://github.com/commaai/openpilot/blob/79122ae66230cac5457e09011435373967d0e1bf/selfdrive/test/test_car_models.py#L452) on new car ports until they have a route in the tests.

pd0wm, 2020-01-20:
Yes. I'm aware.

We will try to get the list of public routes as complete as possible soon. I'll post on discord to see if people want to "donate" a route.

heatnation, 2020-01-20:
I want to donate my route on my Rav4 hybrid 2016 model dsu disconnected

pd0wm, 2020-01-20:
> I want to donate my route on my Rav4 hybrid 2016 model dsu disconnected

If you want to donate a route, please respond with the route id. Example `f1b4c567731f4a1b|2018-06-06--14-43-46`

heatnation, 2020-01-22:
Here is the routes I want to donate.... 
13591d1e8bb0f230|2020-01-21--21-07-14--0 -RAV4 hybrid 2016

arne182, 2020-01-22:
bdfdfa3621d4382e|2020-01-21--14-31-11 -Toyota RAV4 Hybrid 2018

tiki-bug, 2020-02-06:
0a96f86fcfe35964|2020-02-05--07-25-51--14 - 2020 Honda Odyssey EX-L

arne182, 2020-02-06:
32a7df20486b0f70|2020-02-06--16-06-50--11 -Toyota RAV4 Hybrid 2018

pd0wm, 2020-03-06:
`hali: @Willem Melching Honda.Pilot route: 07d37d27996096b6|2020-03-04--21-57-27--1`

csouers, 2020-03-06:
0a78dfbacc8504ef|2020-03-04--13-29-55--0
HONDA.CIVIC_BOSCH

jadtec87, 2020-03-06:
6cdecc4728d4af37|2020-02-23--15-44-18 Toyota Camry 2018

Kiaoptima, 2020-03-06:
7653b2bce7bcfdaa¦7C2020-03-04--16

Kia Optima 2016

cuongluong, 2020-03-06:
702999b9d9f2da2b|2020-03-05--15-45-01

TOYOTA CAMRY 2019

pd0wm, 2020-03-06:
@Kiaoptima Do you have a shorter route? Can I use `7653b2bce7bcfdaa|2020-03-04--15-34-32`

thedp79, 2020-03-06:
420a8e183f1aed48|2020-03-05--07-15-29--0

2017 Pacifica Hybrid

Kiaoptima, 2020-03-06:
> @Kiaoptima Do you have a shorter route? Can I use `7653b2bce7bcfdaa|2020-03-04--15-34-32`?

Yes of course go ahead and use it! 🙂

gregjhogan, 2020-03-06:
2017 cr-v (5g):  `1ad763dd22ef1a0e|2020-02-29--18-37-03--3`

pd0wm, 2020-03-07:
Thanks for all the routes! Adding them to the test right now.

TomDz1, 2020-03-08:
Honda.Pilot_2019: 22affd6c545d985e|2020-03-08--01-08-09--0

Kamzu, 2020-03-08:
4556e5fb8b42b033|2020-03-07--10-52-09--1 CHRYSLER.PACIFICA_2018,

pd0wm, 2020-03-11:
@Kamzu Do you have a shorter route that is also completely uploaded?

VirtuallyChris, 2020-10-09:
f34a60d68d83b1e5|2020-10-06--14-35-55
Acura RDX route from quocuong on Discord

barghe, 2021-04-07:
32b4dc46430470bd | 2021-04-06--18-11-23 | Hyundai CADENZA(K7) 2017

j-vanetten, 2021-05-06:
2018 JEEP GRAND CHEROKEE TRAILHAWK
0c94aa1e1296d7c6|2021-05-05--19-48-37--2

archlyg, 2021-05-18:
c5ac319aa9583f83|2021-05-18--09-42-45--0 | Hyundai Elantra 2017 Limited Ultimate

ShaneSmiskol, 2021-05-28:
Updated list of missing models for anyone interested:

Hyundai/Kia/Genesis:
- [ ] Elantra
- [ ] Elantra GT I30
- [ ] Santa Fe
- [ ] Genesis G90
- [ ] Kia Forte
- [ ] Kia Optima Hybrid
- [ ] Kona EV

Chrysler:
- [ ] Jeep Cherokee 2019
- [ ] Pacifica 2018
- [ ] Pacifica 2018 Hybrid
- [ ] Pacifica 2020

GM:
- [ ] Cadillac ATS
- [ ] Holden Astra
- [ ] Malibu

ShaneSmiskol, 2021-05-28:
Yes, if it is running stock, that would work. Post a route name where you are driving engaged and we can add it!

archlyg, 2021-05-28:
Cool! I don't know if this works or not. I was having an issue that openpilot disengages every time I turn on the stock ACC. It shows up the control mismatch warning and the LKAS loses signal. 

Here's the route:
c5ac319aa9583f83|2021-05-18--09-42-45--0 | Hyundai Elantra 2017 Limited Ultimate

ShaneSmiskol, 2021-05-29:
I don't know if we want to add that route as a test, but just so you know your logs aren't uploading. Make sure upload raw logs is toggled on in settings if you want that information uploaded

adeebshihadeh, 2021-06-01:
Closing, looks like there's no active users for the rest of the missing cars.

closed 2021-06-01 by adeebshihadeh

archlyg, 2021-06-01:
@ShaneSmiskol Sure! Is that done through comma 2 or through the explorer portal? I thought it was uploaded.

archlyg, 2021-06-02:
@ShaneSmiskol I think I uploaded the Raw Logs this time. I just updated the software to the 0.8.4 version today and the problem persists.

c5ac319aa9583f83|2021-06-01--18-18-31 | Hyundai Elantra 2017 Limited Ultimate

ShaneSmiskol, 2021-06-02:
I saw the logs uploaded now, thanks, I'll take a look!

archlyg, 2021-06-02:
cool! thanks so much! hope the problem can be solved!

ShaneSmiskol, 2021-06-02:
@archlyg Can you test this branch out and see if it resolves the controls mismatches? https://github.com/commaai/openpilot/pull/21108